stm32duino/STM32SD #2: FatFs 1.4.1 & Core 1.3.0 - PR #269 Fails to compile

BennehBoy, 2018-06-25:
I'm trying to add SD support to the BLACK407VE variant.

To do this I have copied the sdconf.h & ffconf.h from DISCO_F746VG's variant folder.

Here are the compilation errors (this is with the my own sketch but I also get similar with the Datalogger.ino example).  Datalogger.ino compiles fine against DISCO_F746VG board.

I fully expect that I've done something stupid, please educate me :)

```Build options changed, rebuilding all
In file included from C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/Sd2Card.h:40:0,

                 from C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/STM32SD.h:22,

                 from C:\Users\XXXX\Documents\Arduino\LeeDuino\LRduino---F407VET6-STM\LRduino---F407VET6-STM.ino:8:

C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:51:40: error: variable or field 'BSP_SD_GetCardInfo' declared void

 #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef

                                        ^

C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:82:28: note: in expansion of macro 'HAL_SD_CardInfoTypedef'

 void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);

                            ^~~~~~~~~~~~~~~~~~~~~~

C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:51:40: error: 'HAL_SD_CardInfoTypeDef' was not declared in this scope

 #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef

                                        ^

C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:82:28: note: in expansion of macro 'HAL_SD_CardInfoTypedef'

 void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);

                            ^~~~~~~~~~~~~~~~~~~~~~

C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:82:52: error: 'CardInfo' was not declared in this scope

 void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);

                                                    ^~~~~~~~

C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:87:24: error: variable or field 'BSP_SD_MspInit' declared void

 void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);

                        ^~~~~~~~~~~~~~~~

C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:87:24: error: 'SD_HandleTypeDef' was not declared in this scope

C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:87:42: error: 'hsd' was not declared in this scope

 void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);

                                          ^~~

C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:87:47: error: expected primary-expression before 'void'

 void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);

                                               ^~~~

C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:88:31: error: variable or field 'BSP_SD_Detect_MspInit' declared void

 void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);

                               ^~~~~~~~~~~~~~~~

C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:88:31: error: 'SD_HandleTypeDef' was not declared in this scope

C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:88:49: error: 'hsd' was not declared in this scope

 void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);

                                                 ^~~

C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:88:54: error: expected primary-expression before 'void'

 void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);

                                                      ^~~~

C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:89:26: error: variable or field 'BSP_SD_MspDeInit' declared void

 void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);

                          ^~~~~~~~~~~~~~~~

C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:89:26: error: 'SD_HandleTypeDef' was not declared in this scope

C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:89:44: error: 'hsd' was not declared in this scope

 void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);

                                            ^~~

C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:89:49: error: expected primary-expression before 'void'

 void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);

                                                 ^~~~

C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:51:40: error: 'HAL_SD_CardInfoTypeDef' does not name a type

 #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef

                                        ^

C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:57:21: note: in expansion of macro 'HAL_SD_CardInfoTypedef'

 #define SD_CardInfo HAL_SD_CardInfoTypedef

                     ^~~~~~~~~~~~~~~~~~~~~~

C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/Sd2Card.h:66:3: note: in expansion of macro 'SD_CardInfo'

   SD_CardInfo _SdCardInfo;

   ^~~~~~~~~~~

In file included from C:\Users\XXXX\Documents\Arduino\libraries\FatFs\src/ff.h:27:0,

                 from C:\Users\XXXX\Documents\Arduino\libraries\FatFs\src/ff_gen_drv.h:57,

                 from C:\Users\XXXX\Documents\Arduino\libraries\FatFs\src/FatFs.h:42,

                 from C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/SdFatFs.h:43,

                 from C:\Users\XXXX\Documents\Arduino\libraries\STM32duino_STM32SD\src/STM32SD.h:23,

                 from C:\Users\XXXX\Documents\Arduino\LeeDuino\LRduino---F407VET6-STM\LRduino---F407VET6-STM.ino:8:

C:\Users\XXXX\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.3.0\variants\BLACK_F407VE/ffconf.h:263:33: error: 'osSemaphoreId' does not name a type

 #define _SYNC_t                 osSemaphoreId

                                 ^

C:\Users\XXXX\Documents\Arduino\libraries\FatFs\src/ff.h:95:2: note: in expansion of macro '_SYNC_t'

  _SYNC_t sobj;   /* Identifier of sync object */

  ^~~~~~~

C:\Users\XXXX\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.3.0\variants\BLACK_F407VE/ffconf.h:263:33: error: 'osSemaphoreId' has not been declared

 #define _SYNC_t                 osSemaphoreId

                                 ^

C:\Users\XXXX\Documents\Arduino\libraries\FatFs\src/ff.h:293:31: note: in expansion of macro '_SYNC_t'

 int ff_cre_syncobj (BYTE vol, _SYNC_t* sobj); /* Create a sync object */

                               ^~~~~~~

C:\Users\XXXX\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.3.0\variants\BLACK_F407VE/ffconf.h:263:33: error: 'osSemaphoreId' was not declared in this scope

 #define _SYNC_t                 osSemaphoreId

                                 ^

C:\Users\XXXX\Documents\Arduino\libraries\FatFs\src/ff.h:294:19: note: in expansion of macro '_SYNC_t'

 int ff_req_grant (_SYNC_t sobj);    /* Lock sync object */

                   ^~~~~~~

C:\Users\XXXX\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.3.0\variants\BLACK_F407VE/ffconf.h:263:33: error: variable or field 'ff_rel_grant' declared void

 #define _SYNC_t                 osSemaphoreId

                                 ^

C:\Users\XXXX\Documents\Arduino\libraries\FatFs\src/ff.h:295:20: note: in expansion of macro '_SYNC_t'

 void ff_rel_grant (_SYNC_t sobj);    /* Unlock sync object */

                    ^~~~~~~

C:\Users\XXXX\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.3.0\variants\BLACK_F407VE/ffconf.h:263:33: error: 'osSemaphoreId' was not declared in this scope

 #define _SYNC_t                 osSemaphoreId

                                 ^

C:\Users\XXXX\Documents\Arduino\libraries\FatFs\src/ff.h:295:20: note: in expansion of macro '_SYNC_t'

 void ff_rel_grant (_SYNC_t sobj);    /* Unlock sync object */

                    ^~~~~~~

C:\Users\XXXX\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.3.0\variants\BLACK_F407VE/ffconf.h:263:33: error: 'osSemaphoreId' was not declared in this scope

 #define _SYNC_t                 osSemaphoreId

                                 ^

C:\Users\XXXX\Documents\Arduino\libraries\FatFs\src/ff.h:296:21: note: in expansion of macro '_SYNC_t'

 int ff_del_syncobj (_SYNC_t sobj);    /* Delete a sync object */

                     ^~~~~~~

exit status 1
Error compiling for board Black F407VE.```

BennehBoy, 2018-06-26:
OK so I got a bit further, some silly mistakes by me...

Actually enabled the SD device in stm32f4xx_hal_conf.h - DOH!

Next - I used the ffconf.h from DISCO_F476NG, the SD device on that board is SDMMC1, I switched this out for SDIO after looking in the CMSIS device define in stm32f407xx.h

But now I'm out of my depth because I don't know how to configure the other settings in sdconf.h

ie everything except SD_INSTANCE below:
```
/*SD-detect signal */
#define SD_DETECT_PIN                        GPIO_PIN_13
#define SD_DETECT_GPIO_PORT                  GPIOC
#define SD_DETECT_GPIO_CLK_ENABLE()          __HAL_RCC_GPIOC_CLK_ENABLE()
#define SD_DETECT_GPIO_CLK_DISABLE()         __HAL_RCC_GPIOC_CLK_DISABLE()
#define SD_DETECT_EXTI_IRQn                  EXTI15_10_IRQn

/* Definition for BSP SD */
#define SD_INSTANCE                          SDIO
 
#define SD_CLK_ENABLE                        __HAL_RCC_SDMMC1_CLK_ENABLE
#define SD_CLK_DISABLE                       __HAL_RCC_SDMMC1_CLK_DISABLE
#define SD_CLK_EDGE                          SDMMC_CLOCK_EDGE_RISING
#define SD_CLK_BYPASS                        SDMMC_CLOCK_BYPASS_DISABLE
#define SD_CLK_PWR_SAVE                      SDMMC_CLOCK_POWER_SAVE_DISABLE
#define SD_BUS_WIDE_1B                       SDMMC_BUS_WIDE_1B
#define SD_BUS_WIDE_4B                       SDMMC_BUS_WIDE_4B
#define SD_HW_FLOW_CTRL                      SDMMC_HARDWARE_FLOW_CONTROL_DISABLE
#define SD_CLK_DIV                           SDMMC_TRANSFER_CLK_DIV

/* Definition for MSP SD */
#define SD_AF                                GPIO_AF12_SDMMC1
```

Is there some tooling/docs I need to be using to pull out this magic?

BennehBoy, 2018-06-27:
I had a look at the GENERIC implementation of SDIO and it appears to use the same mechanism as the F7 implementation, but I can only see the above definitions for F7 in CMSIS  (specifically stm32f7xx_ll_sdmmc.h).  This is all a bit outside of my knowledge now - lots of reading required no doubt

BennehBoy, 2018-08-01:
@fpistm what's the current status of this?  Still awaiting some time?

fpistm, 2018-08-01:
@BennehBoy 
As I said it require an update. If I well remembered, this is mainly a switch issue. F4 HAL SD moved to the same F7 implementation. Unfortunately, I did not get time to update it. I will try to have a look if this is not huge.

BennehBoy, 2018-08-03:
If I get time I'll have a tinker and drop the newer HAL definitions into the core - but I expect there are likely other downstream changes required to accommodate those.  C'est la vie!

fpistm, 2018-09-17:
Hi @BennehBoy 
I've made a quick test and it is ok.
In fact no update needed related to HAL (I confused myself between F4 and L4 sorry :().
So to have it working:
1 - Enable `HAL_SD_MODULE_ENABLED`
2 - Add the `ffconf.h` from F7
3 - Add the `sd_conf.h` with the updated define:

```
/**
  ******************************************************************************
  * @file    sd_conf.h
  * @brief   This file contains hw config for SD
  *******************************************************************************
  */

/* Define to prevent recursive inclusion -------------------------------------*/
#ifndef __SD_CONF_H
#define __SD_CONF_H

#ifdef __cplusplus
 extern "C" {
#endif

/* Includes ------------------------------------------------------------------*/
#include "stm32_def.h"

#define SD_DATATIMEOUT           ((uint32_t)100000000)

/*SD-detect signal */
#define SD_DETECT_PIN                        GPIO_PIN_13
#define SD_DETECT_GPIO_PORT                  GPIOC
#define SD_DETECT_GPIO_CLK_ENABLE()          __HAL_RCC_GPIOC_CLK_ENABLE()
#define SD_DETECT_GPIO_CLK_DISABLE()         __HAL_RCC_GPIOC_CLK_DISABLE()
#define SD_DETECT_EXTI_IRQn                  EXTI15_10_IRQn

/* Definition for BSP SD */
#define SD_INSTANCE                          SDIO

#define SD_CLK_ENABLE                        __HAL_RCC_SDMMC1_CLK_ENABLE
#define SD_CLK_DISABLE                       __HAL_RCC_SDMMC1_CLK_DISABLE
#define SD_CLK_EDGE                          SDIO_CLOCK_EDGE_RISING
#define SD_CLK_BYPASS                        SDIO_CLOCK_BYPASS_DISABLE
#define SD_CLK_PWR_SAVE                      SDIO_CLOCK_POWER_SAVE_DISABLE
#define SD_BUS_WIDE_1B                       SDIO_BUS_WIDE_1B
#define SD_BUS_WIDE_4B                       SDIO_BUS_WIDE_4B
#define SD_HW_FLOW_CTRL                      SDIO_HARDWARE_FLOW_CONTROL_DISABLE
#define SD_CLK_DIV                           SDIO_TRANSFER_CLK_DIV

/* Definition for MSP SD */
#define SD_AF                                GPIO_AF12_SDIO

#ifdef __cplusplus
}
#endif

#endif /* __SD_CONF_H */
```

Note that it seems the BlackF7 seems do not have the card detect connected.
So, all usage  of `SD_DETECT_PIN` in example have to be removed.

fpistm, 2018-09-17:
FYI, I will remove the use of sd_conf.h as this could be added directly in the library. Only the SD detect pins will have to be defined in the variant.h if any

fpistm, 2018-09-19:
Hi @BennehBoy,
Black F407 has been added to the core with SD HAL enabled and will be available in the core 1.3.1:
https://github.com/stm32duino/Arduino_Core_STM32/pull/327

Thanks the next STM32SD library 1.1.0 version including #4, there is no more need to add `sd_conf.h` and `ffconf.h`.
SD is now functional on the Black F407VET.

closed 2018-09-19 by fpistm