stm32duino/STM32SD #5: stm32_def.h

GnuReligion, 2018-09-19:
~/sketchbook/libraries/STM32SD-master/src/ffconf.h:8:23: fatal error: stm32_def.h: No such file or directory
 #include "stm32_def.h"

I cannot locate the stm32_def.h file in STM32SD or FatFs

fpistm, 2018-09-19:
Hi @GnuReligion 
Which core you used ?
`stm32_def.h` is part of the https://github.com/stm32duino/Arduino_Core_STM32
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/cores/arduino/stm32/stm32_def.h

GnuReligion, 2018-09-21:
Thank you, this was my problem.
I was using:
`http://dan.drown.org/stm32duino/package_STM32duino_index.json`
Instead of:
`https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json`

But ... now I have different errors compiling CardInfo.
This is not so important to me now though, as I found it is possible to use the native Arduino SPI code to manipulate SD Cards.

`"/tmp/arduino_build_520014/sketch/CardInfo.ino.cpp" -o "/tmp/arduino_build_520014/sketch/CardInfo.ino.cpp.o"
In file included from ~/sketchbook/libraries/STM32SD-master/src/Sd2Card.h:40:0,
                 from ~/sketchbook/libraries/STM32SD-master/src/STM32SD.h:22,
                 from ~/sketchbook/libraries/STM32SD-master/examples/CardInfo/CardInfo.ino:12:
~/sketchbook/libraries/STM32SD-master/src/bsp_sd.h:85:1: error: 'HAL_SD_TransferStateTypedef' does not name a type
 HAL_SD_TransferStateTypedef BSP_SD_GetStatus(void);
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~
~/sketchbook/libraries/STM32SD-master/src/bsp_sd.h:87:28: error: variable or field 'BSP_SD_GetCardInfo' declared void
 void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);`

fpistm, 2018-09-22:
Which board you used?
This error means the variant you used do not have the `HAL_SD_MODULE_ENABLED` define.

jerlon1, 2018-10-15:
Hi, i have a similar problem.
If I use board "Generic STM32F103 series"  then Board Part Number STM32F103C8" then i get the error related to HAL_SD_TransferStateTypedef'
If I use Board "Generic STM32F103C series" then I get the "stm32_def.h"
Now I am a little confused with the whole IDE set up.

fpistm, 2018-10-15:
This library is only for this core (ST HAL based):
https://github.com/stm32duino/Arduino_Core_STM32
which can be installed thanks:
https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json
With this core and the `Generic STM32F103 series` with Board Part Number` BluePillF103C8`, this is normal as the Bluepill do not have a SD card slot connected to the SDIO interface, that's why the `HAL_SD_MODULE_ENABLED` is not enabled by default for this board.

If you have the Board Part Number STM32F103C8... I guess you use Roger's core:
https://github.com/rogerclarkmelbourne/Arduino_STM32
So this library is not compatible.

fpistm, 2018-10-15:
@GnuReligion any feedback?

fpistm, 2018-10-15:
After reading again the thread, I guess several user try to use this library for SD card connected through SPI while it is for SD Card connected through SDIO interface.
I guess I need to update the README.md and the library.properties files.

GnuReligion, 2018-10-27:
> Which board you used?
> This error means the variant you used do not have the `HAL_SD_MODULE_ENABLED` defined

So sorry for taking so long to reply.  Gmail was sending github notices to spam.

I build my own boards, this one with an stm32f103c8t6, modeled similarly to a Blue Pill. 
It has 128kb Flash though, so it may be a cBt6, in reality ... but this is irrelevant.

Have connected an ICSP-style 6-pin SPI interface to it using the primary SPI pins, PA4-PA7.
The card-reading device I am using is an SPI SD card reader labeled "MH-SD Card Module" from China.

From what I gather here, I need a DIFFERENT kind of SD interface, perhaps using "real" SD and not generic SPI, for this library.  My apologies for being ignorant on the matter.

Was interested in your library because I hear that the SD protocol is much faster.
Guess I need a new boards definition with HAL_SD_MODULE_ENABLED, among other things defined.
Hmm, wondering now, how do I build this interface?

Sorry for wasting your time.

GnuReligion, 2018-10-28:
Finally, I understand after reading this post:  http://www.stm32duino.com/viewtopic.php?t=288
The MCU itself must support a thing called "SDIO" in order to use this library.  Without hardware support, an MCU must have quite a bit of RAM for the protocol, so my stm32f103c8t6 chips will never work.

Have just ordered a couple of STM32F103VCT6 chips though, which do explicitly support SDIO ... so I will be trying your library again, with these, in a few weeks.

closed 2018-10-28 by GnuReligion