stm32duino/STM32SD #8: SD card is not accessing

Rajkumar181, 2018-11-14:
Hi Team,
  I need your support to bring up my board. I trying to interface  with SD card. but it is not communicating.
I tried SD card read/write example. 

error opening test.txt
error opening test.txt
###### End of the SD tests ######

Board - STM32F429zi
SD card - SD MMC

i attached my schematic for reference.     

[image]
[image]

Rajkumar181, 2018-11-16:
Any update why this is not working. Is it API have some issue.

fpistm, 2018-11-16:
Hi,
this is difficult to tell as I can not reproduce.
Could you give more detail:
 * Core version
 * STM32SD library version
 * FatFs library version
 * It seems to be a custom board, so which variant you used ? Is system clock config ok ?
 * Which SD card? Is is a proper FS ?
...
What is the result of CardInfo ?

Rajkumar181, 2018-11-16:
Thanks for your replay,

Rajkumar181, 2018-11-16:
Thanks for your Reply,

1. - Core version -- Arduino Core for STM32 1.4.0
2. - STM32SD library version --  1.1.1
3. - FatFs library version -- 2.0.2
4. - It seems to be a custom board, so which variant you used ?
             yes it is a custom board , And variant is NUCLEO_F429ZI.

 Is system clock config ok ?
                Default clock system.

5. - Which SD card? 
       Make - SanDisk  16GB, Class 10
        Micro SD

    Is a proper FS ?
               yes

fpistm, 2018-11-16:
So this is normal that you cannot open a file:

> Could not find FAT16/FAT32 partition.
> Make sure you've formatted the card

About the default clock config, it assume you have an HSE at 8MHz

Rajkumar181, 2018-11-16:
Yes it is  there.

[image]

fpistm, 2018-11-16:
Ok. Have you try  format your SDCard to a FAT 32 FS?

Rajkumar181, 2018-11-16:
Yes i did that. Same response

fpistm, 2018-11-16:
Unfortunately, I will not be able to help on that seems FatFs does not recognized you FS.
SD init seems weel, so I guess this is not an hardware issue.
Maybe the FATFs config is not correct for your SD (codepage,...)

Rajkumar181, 2018-11-16:
Sorry i didn't change any configuration for SD card.

This configuration is runing now.

https://github.com/stm32duino/FatFs/blob/master/src/ffconf_template.h

fpistm, 2018-11-19:
Unfortunately, I don't know what to suggest.
It seems the SD card properly init then It could not access properly to the FS.
How do you format your SD ?
Have you tried with another SD?

Rajkumar181, 2018-11-19:
I tried SPI line for interface my SD Card. Its working fine on same line. But SDIO interface is not working. 

How do you format your SD ?
I formatted SD Card using FAT32.

Have you tried with another SD?
Yes, I tried 3 SD card.

fpistm, 2018-11-19:
> I formatted SD Card using FAT32.

Ok but which tools you used to do that?

As I said, It's difficult to answer you as I can't reproduce. The best you can do will be to debug on your side to understand why FS is not properly accessed through FatFs. SDIO seems ok as card is init.

Rajkumar181, 2018-11-19:
Ok but which tools you used to do that?
Windows tools

[image]

SDIO init also getting failed.

fpistm, 2018-11-19:
No SD init is ok as Card info told you :
`Card type: SD1`

FatFs limitations:

>     Filesystem type: FAT, FAT32(rev0.0) and exFAT(rev1.0).
>     Number of open files: Unlimited. (depends on available memory)
>     Number of volumes: Upto 10.
>     Volume size: Upto 2 TB at 512 bytes/sector.
>     File size: Upto 4 GB - 1 on FAT volume and virtually unlimited on exFAT volume.
>     Cluster size: Upto 128 sectors on FAT volume and upto 16 MB on exFAT volume.
>     Sector size: 512, 1024, 2048 and 4096 bytes.

So try with lesser Allocation unit size  and do not perform a quick format.

Rajkumar181, 2018-11-19:
Again i try Card info example
While initialization the card, it's taking more time.
Output:   Initializing SD card...initialization failed. Is a card inserted?

[image]

fpistm, 2018-11-19:
> What is the result of CardInfo ?
> Card type: SD1
> Could not find FAT16/FAT32 partition.
> Make sure you've formatted the card

You wrote this, so for me SD card init was OK... Only FS issue.
Now you told me that it is not OK. What has changed between those 2 different answers?

Rajkumar181, 2018-11-19:
What is the result of CardInfo ?
Initializing SD card...initialization failed. Is a card inserted?

Make sure you've formatted the card
Yes I have formatted the card.

Yes Card init is not ok. 
Sorry , Actually init is a problem. That's why i shared my schematics.

fpistm, 2018-11-19:
If I read the history of this issue, I can see init was succeed some times.
In your first post you said you had:
```
error opening test.txt
error opening test.txt
```
This only can happen in the original example when the SD is well initialized.
Same for the result card info.
Now you told this is not ok. How can I support you if your inputs are not ok.
STM32SD + FatFs is working on STM32F4 series and tested thanks the BlackF4 boards.
As it is a custom board, you have to validate your hardware. Maybe you should check black F4 schematics.
Here this is not a place to request schematics review.

Rajkumar181, 2018-11-19:
I check my schematics with BlackF4 board. This ok.

In my first post i did one mistake 
         while (SD.begin(SD_DETECT_PIN))
that's why i got this :- 
    error opening test.txt
    error opening test.txt

Just now i debug  the code , so i founded where is it getting fail.
File name - bsp_sd.c 

In this function:-  uint8_t BSP_SD_Init(void)
 if(HAL_SD_WideBusOperation_Config(&uSdHandle, SD_BUS_WIDE_4B) != SD_OK)
    {          
      sd_state = MSD_ERROR;
    }
    else
    {
      sd_state = MSD_OK;
    }

In this director "HAL_SD_WideBusOperation_Config" function is going to call
system/Drivers/STM32L1xx_HAL_Driver/Src/stm32l1xx_hal_sd.c

fpistm, 2018-11-19:
OK.
but this is system/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_sd.c
Then `HAL_SD_WideBusOperation_Config` is `HAL_SD_ConfigWideBusOperation`
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/system/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_sd.c#L2073

You will have to go further to know what is failed.
`  if(hsd->SdCard.CardType != CARD_SECURED)`  ?
or 
`      errorstate = SD_WideBus_Enable(hsd)`; ?

Rajkumar181, 2018-11-20:
I change  HAL_SD_WideBusOperation_Config to HAL_SD_ConfigWideBusOperation

this is getting fail and returning HAL_ERROR.
errorstate = SD_WideBus_Enable(hsd);

 if(hsd->ErrorCode != HAL_SD_ERROR_NONE)
  {
    /* Clear all the static flags */
    __HAL_SD_CLEAR_FLAG(hsd, SDIO_STATIC_FLAGS);
    hsd->State = HAL_SD_STATE_READY;
    return HAL_ERROR;
  }

fpistm, 2018-11-20:
You do not need to change the function name, it is handled by define to match all series.

So you have to dig into `SD_WideBus_Enable`
https://github.com/stm32duino/Arduino_Core_STM32/blob/1961382fcdca9f87e600017d2e52ff3a24e3d5d3/system/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_sd.c#L2825
to understand why there is an error.

I could not do it for you as I can't reproduce the issue.

fpistm, 2018-11-20:
You do not need to change the function name, it is handled by define to match all series.

So you have to dig into `SD_WideBus_Enable`
https://github.com/stm32duino/Arduino_Core_STM32/blob/1961382fcdca9f87e600017d2e52ff3a24e3d5d3/system/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_sd.c#L2825
to understand why there is an error.

I could not do it for you as I can't reproduce the issue.

Tips:
https://github.com/stm32duino/wiki/wiki/How-to-debug

Rajkumar181, 2018-11-20:
is it Hal library error? or some think else .

fpistm, 2018-11-20:
I don't think it is HAL error. You have to figure out why there is an error  accessing to your SD.
I do not tell this library has no issue or missing feature but as far as I know it works.
You bring up your board, up to you to validate all your board, HAL raised an error and it's probably for a good reason.

Rajkumar181, 2018-11-20:
Ok, I will try with NUCLEO stm32f429zi board.

fpistm, 2019-01-16:
@Rajkumar181 any update on this?

Rajkumar181, 2019-01-18:
Its Resolved. 8 Mhz clock was issue. Thanks for support.

fpistm, 2019-01-18:
Ok. Fine to see you solved this issue

closed 2019-01-18 by fpistm