stm32duino/STM32SD #13: Memory leak?

AnHardt, 2019-01-29:
Take [CardInfo.ino](https://github.com/stm32duino/STM32SD/blob/master/examples/CardInfo/CardInfo.ino)
Make `root` global.
Add
``` cpp
  root.ls(LS_R | LS_DATE | LS_SIZE);
  root.rewindDirectory();
```
to loop.
Compile. Run.

Runs for a while, than finishes with:
```
assertion "_fil != NULL" failed: file "...\Arduino\libraries\STM32duino_STM32SD\src\SD.cpp", line 218, function: File::File(const char*)
```

fpistm, 2019-01-29:
Could you be more precise to be able to reproduce, please.
Boards ?
SD info + number of files ?
Exact sketch you used?

AnHardt, 2019-01-29:
Hardware: Plain Black F407VE with 2MB SD card. Maximum 3 directory levels.
Every software up to date from today. Arduino 1.8.8.
Generic STM32F4 series
Black F407VE
everything else at the defaults

If you really want the complete sketch - here it is.
``` cpp
/*
  SD card test

 This example shows how use the utility libraries on which the'
 SD library is based in order to get info about your SD card.
 Very useful for testing a card when you're not sure whether its working or not.

  * SD card attached

 */
// include the SD library:
#include <STM32SD.h>

// If SD card slot has no detect pin then define it as SD_DETECT_NONE
// to ignore it. One other option is to call 'card.init()' without parameter.
#ifndef SD_DETECT_PIN
#define SD_DETECT_PIN SD_DETECT_NONE
#endif

Sd2Card card;
SdFatFs fatFs;

File root;  // root made global ***************************************************************

void setup()
{
  bool disp = false;
  // Open serial communications and wait for port to open:
  Serial.begin(115200);

  while (!Serial);
  Serial.print("\nInitializing SD card...");
  while(!card.init(SD_DETECT_PIN)) {
    if (!disp) {
      Serial.println("initialization failed. Is a card inserted?");
      disp = true;
    }
    delay(10);
  }

  Serial.println("A card is present.");

  // print the type of card
  Serial.print("\nCard type: ");
  switch (card.type()) {
    case SD_CARD_TYPE_SD1:
      Serial.println("SD1");
      break;
    case SD_CARD_TYPE_SD2:
      Serial.println("SD2");
      break;
    case SD_CARD_TYPE_SDHC:
      Serial.println("SDHC");
      break;
    default:
      Serial.println("Unknown");
  }

  // Now we will try to open the 'volume'/'partition' - it should be FAT16 or FAT32
  if (!fatFs.init()) {
    Serial.println("Could not find FAT16/FAT32 partition.\nMake sure you've formatted the card");
    return;
  }

  // print the type and size of the first FAT-type volume
  uint64_t volumesize;
  Serial.print("\nVolume type is FAT");
  Serial.println(fatFs.fatType(), DEC);
  Serial.println();

  volumesize = fatFs.blocksPerCluster();    // clusters are collections of blocks
  volumesize *= fatFs.clusterCount();       // we'll have a lot of clusters
  volumesize *= 512;                        // SD card blocks are always 512 bytes
  Serial.print("Volume size (bytes): ");
  Serial.println(volumesize);
  Serial.print("Volume size (Kbytes): ");
  volumesize /= 1024;
  Serial.println(volumesize);
  Serial.print("Volume size (Mbytes): ");
  volumesize /= 1024;
  Serial.println(volumesize);

  Serial.println("\nFiles found on the card (name, date and size in bytes): ");
  root = SD.openRoot(); // root made global ***************************************************************

  // list all files in the card with date and size
  root.ls(LS_R | LS_DATE | LS_SIZE);
  Serial.println("###### End of the SD tests ######");
}

void loop(void) {
// ******************************************* added *************************************************
  root.ls(LS_R | LS_DATE | LS_SIZE);
  root.rewindDirectory();
// ******************************************* added *************************************************
}
```

fpistm, 2019-01-29:
Ok thanks for the feedback.
As I think you use the `rewindDirectory()` and I've already open an issue about that #12 .

fpistm, 2019-04-23:
After some investigations, I've found the issue.
I will push a fix for that.

closed 2019-04-23 by fpistm

fpistm, 2019-04-23:
@AnHardt 
Running all day without any issue.
Do not hesitate if you met any issue.

fpistm, 2019-04-24:
And running all the night ;)

commit 1868af1f (2019-04-23): Free file object structure (FIL) member

Fix #13