stm32duino/STM32SD #16: Version 1.6.1 Issue

zpowellman, 2019-08-18:
Trying to compile full.ino on black stm32F407VE board.
Get the following message:
#error "This library version required a STM32 core version > 1.6.1.\
I'm using Version 1.6.1 of the STM Core

Any ideas?
Thanks,
Dan

zpowellman, 2019-08-18:
OOPs!  Just saw the > symbol.  I thought 1.6.1 was the latest.

fpistm, 2019-08-18:
Right,install the latest release thanks the library manager not the zip of the master or try the core master.

zpowellman, 2019-08-18:
Thanks for the response.

closed 2019-08-18 by zpowellman

zpowellman, 2019-08-18:
Frederic,

Thanks for the reply.  Was this library intended to be use with SPI or SDIO.  I’m using a STM32F407VET6 board?

Thanks again, the compile and download was successful but the app can’t find the card.

Dan

fpistm, 2019-08-18:
SDIO or SDMMC.
For SPI use the SD library from arduino.