stm32duino/STM32SD #17: CardInfo.ino: Example cannot be compiled, trying with blue pill (STM32F103CBT6, 128k)

rtek1000, 2019-10-16:
How can I make code compile?

Arduino platform not famous for not having to edit external files in the example code?

Thank you.

In file included from C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/Sd2Card.h:40,
                 from C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/STM32SD.h:22,
                 from C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\examples\CardInfo\CardInfo.ino:12:
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:58:40: error: variable or field 'BSP_SD_GetCardInfo' declared void
 #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef
                                        ^~~~~~~~~~~~~~~~~~~~~~
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:115:28: note: in expansion of macro 'HAL_SD_CardInfoTypedef'
 void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);
                            ^~~~~~~~~~~~~~~~~~~~~~
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:58:40: error: 'HAL_SD_CardInfoTypeDef' was not declared in this scope
 #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef
                                        ^~~~~~~~~~~~~~~~~~~~~~
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:115:28: note: in expansion of macro 'HAL_SD_CardInfoTypedef'
 void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);
                            ^~~~~~~~~~~~~~~~~~~~~~
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:58:40: note: suggested alternative: 'HAL_SD_CardInfoTypedef'
 #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef
                                        ^~~~~~~~~~~~~~~~~~~~~~
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:115:28: note: in expansion of macro 'HAL_SD_CardInfoTypedef'
 void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);
                            ^~~~~~~~~~~~~~~~~~~~~~
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:115:52: error: 'CardInfo' was not declared in this scope
 void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);
                                                    ^~~~~~~~
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:115:52: note: suggested alternative: 'SD_CardInfo'
 void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);
                                                    ^~~~~~~~
                                                    SD_CardInfo
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:120:24: error: variable or field 'BSP_SD_MspInit' declared void
 void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);
                        ^~~~~~~~~~~~~~~~
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:120:24: error: 'SD_HandleTypeDef' was not declared in this scope
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:120:24: note: suggested alternative: 'SPI_HandleTypeDef'
 void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);
                        ^~~~~~~~~~~~~~~~
                        SPI_HandleTypeDef
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:120:42: error: 'hsd' was not declared in this scope
 void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);
                                          ^~~
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:120:47: error: expected primary-expression before 'void'
 void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);
                                               ^~~~
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:121:31: error: variable or field 'BSP_SD_Detect_MspInit' declared void
 void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);
                               ^~~~~~~~~~~~~~~~
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:121:31: error: 'SD_HandleTypeDef' was not declared in this scope
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:121:31: note: suggested alternative: 'SPI_HandleTypeDef'
 void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);
                               ^~~~~~~~~~~~~~~~
                               SPI_HandleTypeDef
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:121:49: error: 'hsd' was not declared in this scope
 void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);
                                                 ^~~
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:121:54: error: expected primary-expression before 'void'
 void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);
                                                      ^~~~
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:122:26: error: variable or field 'BSP_SD_MspDeInit' declared void
 void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);
                          ^~~~~~~~~~~~~~~~
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:122:26: error: 'SD_HandleTypeDef' was not declared in this scope
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:122:26: note: suggested alternative: 'SPI_HandleTypeDef'
 void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);
                          ^~~~~~~~~~~~~~~~
                          SPI_HandleTypeDef
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:122:44: error: 'hsd' was not declared in this scope
 void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);
                                            ^~~
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:122:49: error: expected primary-expression before 'void'
 void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);
                                                 ^~~~
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:58:40: error: 'HAL_SD_CardInfoTypeDef' does not name a type; did you mean 'HAL_SD_CardInfoTypedef'?
 #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef
                                        ^~~~~~~~~~~~~~~~~~~~~~
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/bsp_sd.h:64:21: note: in expansion of macro 'HAL_SD_CardInfoTypedef'
 #define SD_CardInfo HAL_SD_CardInfoTypedef
                     ^~~~~~~~~~~~~~~~~~~~~~
C:\Arduino IDE\arduino-1.8.10-windows-portable\portable\sketchbook\libraries\STM32SD\src/Sd2Card.h:62:5: note: in expansion of macro 'SD_CardInfo'
     SD_CardInfo _SdCardInfo;
     ^~~~~~~~~~~

fpistm, 2019-10-16:
Hi @rtek1000 
That is normal as the BluepPill does not have an SD card slot connected to SDIO, the HAL SD module is not enabled by default.
Finally, STM32F103CBT6 does not have SD...
I guess you try a SD card slot connected through SPI. In that case use default SD library from Arduino.

rtek1000, 2019-10-16:
> This file provides a set of functions needed to manage the SD card ... that is used with the STM32 Nucleo board through SPI interface

Source: https://github.com/eleciawhite/STM32Cube/blob/master/STM32Cube_FW_F3_V1.3.0/Drivers/BSP/Adafruit_Shield/stm32_adafruit_sd.c