stm32duino/STM32SD #18: Cannot compile for Nucleo-64 STM32L452

Bambofy, 2019-10-24:
Hi, i'm trying to use this library and this is the error it reports when trying to compile.
````
In file included from C:\Users\Richa\Documents\GitHub\Remote-Train-Sensor\STM32\FIRST_SKETCH/Sd2Card.h:40,
                 from C:\Users\Richa\Documents\GitHub\Remote-Train-Sensor\STM32\FIRST_SKETCH/STM32SD.h:22,
                 from C:\Users\Richa\Documents\GitHub\Remote-Train-Sensor\STM32\FIRST_SKETCH\FIRST_SKETCH.ino:1:
bsp_sd.h:115:28: error: variable or field 'BSP_SD_GetCardInfo' declared void
 void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);
                            ^~~~~~~~~~~~~~~~~~~~~~
bsp_sd.h:115:28: error: 'HAL_SD_CardInfoTypedef' was not declared in this scope
C:\Users\Richa\Documents\GitHub\Remote-Train-Sensor\STM32\FIRST_SKETCH/bsp_sd.h:115:28: note: suggested alternative: 'HAL_SD_CardCIDTypedef'
 void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);
                            ^~~~~~~~~~~~~~~~~~~~~~
                            HAL_SD_CardCIDTypedef
bsp_sd.h:115:52: error: 'CardInfo' was not declared in this scope
 void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);
                                                    ^~~~~~~~
C:\Users\Richa\Documents\GitHub\Remote-Train-Sensor\STM32\FIRST_SKETCH/bsp_sd.h:115:52: note: suggested alternative: 'SD_CardInfo'
 void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);
                                                    ^~~~~~~~
                                                    SD_CardInfo
bsp_sd.h:120:24: error: variable or field 'BSP_SD_MspInit' declared void
 void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);
                        ^~~~~~~~~~~~~~~~
bsp_sd.h:120:24: error: 'SD_HandleTypeDef' was not declared in this scope
C:\Users\Richa\Documents\GitHub\Remote-Train-Sensor\STM32\FIRST_SKETCH/bsp_sd.h:120:24: note: suggested alternative: 'SPI_HandleTypeDef'
 void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);
                        ^~~~~~~~~~~~~~~~
                        SPI_HandleTypeDef
bsp_sd.h:120:42: error: 'hsd' was not declared in this scope
 void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);
                                          ^~~
bsp_sd.h:120:47: error: expected primary-expression before 'void'
 void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);
                                               ^~~~
bsp_sd.h:121:31: error: variable or field 'BSP_SD_Detect_MspInit' declared void
 void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);
                               ^~~~~~~~~~~~~~~~
bsp_sd.h:121:31: error: 'SD_HandleTypeDef' was not declared in this scope
C:\Users\Richa\Documents\GitHub\Remote-Train-Sensor\STM32\FIRST_SKETCH/bsp_sd.h:121:31: note: suggested alternative: 'SPI_HandleTypeDef'
 void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);
                               ^~~~~~~~~~~~~~~~
                               SPI_HandleTypeDef
bsp_sd.h:121:49: error: 'hsd' was not declared in this scope
 void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);
                                                 ^~~
bsp_sd.h:121:54: error: expected primary-expression before 'void'
 void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);
                                                      ^~~~
bsp_sd.h:122:26: error: variable or field 'BSP_SD_MspDeInit' declared void
 void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);
                          ^~~~~~~~~~~~~~~~
bsp_sd.h:122:26: error: 'SD_HandleTypeDef' was not declared in this scope
C:\Users\Richa\Documents\GitHub\Remote-Train-Sensor\STM32\FIRST_SKETCH/bsp_sd.h:122:26: note: suggested alternative: 'SPI_HandleTypeDef'
 void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);
                          ^~~~~~~~~~~~~~~~
                          SPI_HandleTypeDef
bsp_sd.h:122:44: error: 'hsd' was not declared in this scope
 void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);
                                            ^~~
bsp_sd.h:122:49: error: expected primary-expression before 'void'
 void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);
                                                 ^~~~
bsp_sd.h:64:21: error: 'HAL_SD_CardInfoTypedef' does not name a type; did you mean 'HAL_SD_CardCIDTypedef'?
 #define SD_CardInfo HAL_SD_CardInfoTypedef
                     ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\Richa\Documents\GitHub\Remote-Train-Sensor\STM32\FIRST_SKETCH/Sd2Card.h:62:5: note: in expansion of macro 'SD_CardInfo'
     SD_CardInfo _SdCardInfo;
     ^~~~~~~~~~~
Multiple libraries were found for "FatFs.h"
 Used: C:\Users\Richa\Documents\Arduino\libraries\FatFs-master
Using library FatFs-master at version 2.0.3 in folder: C:\Users\Richa\Documents\Arduino\libraries\FatFs-master 
exit status 1
variable or field 'BSP_SD_GetCardInfo' declared void
```

Bambofy, 2019-10-24:
Sketch is:

`#include <STM32SD.h>

// If SD card slot has no detect pin then define it as SD_DETECT_NONE
// to ignore it. One other option is to call 'card.init()' without parameter.
#ifndef SD_DETECT_PIN
#define SD_DETECT_PIN SD_DETECT_NONE
#endif

Sd2Card card;
SdFatFs fatFs;

const int chipSelect = 10;

void setup() {
Serial.begin(9600);

  Serial.print("\nInitializing SD card...");
  while(!card.init(10)) {
      Serial.println("initialization failed. Is a card inserted?");
    delay(10);
  }

  Serial.println("A card is present.");
}

void loop() {
}`

fpistm, 2019-10-24:
There is no sd card slot per defaut on this board so by default HAL SD module is not enabled. 
Note: this library is not for SD over SPI, for this type of SD shield you have to use default Arduino SD library.

fpistm, 2019-10-24:
I guess I can close it as you open an issue for SD with SPI in the core .

closed 2019-10-24 by fpistm

Bambofy, 2019-10-24:
Thank you, sorry this libraries name fooled me 😆