stm32duino/STM32SD #19: STM32F103 series

zpowellman, 2019-10-26:
It appears this library won't function for the STM32F103 Series.  Is that correct?  I'm using the Super Blue Pill board.
Thanks
Dan

fpistm, 2019-10-26:
Hi, what do you mean by super BP?
I never tested for a F1 but this should work. If not then this would require a fix.

zpowellman, 2019-10-26:
Hi. Thanks for responding. This is the super blue pill:
https://medium.com/swlh/super-blue-pill-like-stm32-blue-pill-but-better-6d341d9347da
Very versatile little board.  SD demo code won't compile for F1 Series.  I can send more info if you'd like?
Thanks again
Dan

fpistm, 2019-10-26:
No, with info you provided that's enough.
On this board this is a SD card over SPI. Moreover the mcu does not have SDIO.
So, this is normal this doesn't build, simply  use the Arduino SD library.
This library is for MCU with card slot connected to SDIO or SDMMC.

zpowellman, 2019-10-26:
Thanks a lot

closed 2019-10-26 by fpistm