stm32duino/STM32SD #21: Couldn't build as per Log below

3nd0y, 2020-03-21:
> *****************************************************************************************************************************************
> Obsolete PIO Core v4.2.1 is used (previous was 4.3.1)
> Please remove multiple PIO Cores from a system:
> https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
> *****************************************************************************************************************************************
> Processing bluepill_f103c8 (platform: ststm32; board: bluepill_f103c8; framework: arduino)
> -----------------------------------------------------------------------------------------------------------------------------------------
> Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/bluepill_f103c8.html
> PLATFORM: ST STM32 6.0.0 > BluePill F103C8
> HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
> DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
> PACKAGES:
>  - framework-arduinoststm32 3.10700.191028 (1.7.0)
>  - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
> LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ soft
> Found 22 compatible libraries
> Scanning dependencies...
> Dependency Graph
> |-- <LiquidCrystal_I2C>
> |   |-- <Wire> 1.0
> |-- <SPI> 1.0
> |-- <DS1307RTC>
> |   |-- <Time> 1.6
> |   |-- <Wire> 1.0
> |-- <STM32duino STM32SD> 1.2.2
> |   |-- <FatFs> 2.0.3
> |-- <Wire> 1.0
> |-- <Time> 1.6
> Building in release mode
> Compiling .pio\build\bluepill_f103c8\src\main.cpp.o
> Compiling .pio\build\bluepill_f103c8\libb76\FatFs_ID1805\diskio.c.o
> Compiling .pio\build\bluepill_f103c8\libb76\FatFs_ID1805\drivers\sd_diskio.c.o
> Compiling .pio\build\bluepill_f103c8\libb76\FatFs_ID1805\ff.c.o
> Compiling .pio\build\bluepill_f103c8\libb76\FatFs_ID1805\ff_gen_drv.c.o
> Compiling .pio\build\bluepill_f103c8\libb76\FatFs_ID1805\option\cc932.c.o
> Compiling .pio\build\bluepill_f103c8\libb76\FatFs_ID1805\option\cc936.c.o
> Compiling .pio\build\bluepill_f103c8\libb76\FatFs_ID1805\option\cc949.c.o
> In file included from C:\Users\fmukhtarif\.platformio\lib\STM32duino STM32SD_ID1810\src/ffconf.h:9:0,
>                  from c:\users\fmukhtarif\.platformio\lib\fatfs_id1805\src\ff.h:29,
>                  from C:\Users\fmukhtarif\.platformio\lib\FatFs_ID1805\src\option\cc932.c:6:
> C:\Users\fmukhtarif\.platformio\lib\STM32duino STM32SD_ID1810\src/bsp_sd.h:58:40: error: unknown type name 'HAL_SD_CardInfoTypeDef'; did you mean 'HAL_SD_CardInfoTypedef'?
>  #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef
>                                         ^
> C:\Users\fmukhtarif\.platformio\lib\STM32duino STM32SD_ID1810\src/bsp_sd.h:115:28: note: in expansion of macro 'HAL_SD_CardInfoTypedef'
>  void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);
>                             ^~~~~~~~~~~~~~~~~~~~~~
> C:\Users\fmukhtarif\.platformio\lib\STM32duino STM32SD_ID1810\src/bsp_sd.h:120:24: error: unknown type name 'SD_HandleTypeDef'; did you mean 'SPI_HandleTypeDef'?
>  void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);
>                         ^~~~~~~~~~~~~~~~
>                         SPI_HandleTypeDef
> C:\Users\fmukhtarif\.platformio\lib\STM32duino STM32SD_ID1810\src/bsp_sd.h:121:31: error: unknown type name 'SD_HandleTypeDef'; did you mean 'SPI_HandleTypeDef'?
>  void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);
>                                ^~~~~~~~~~~~~~~~
>                                SPI_HandleTypeDef
> C:\Users\fmukhtarif\.platformio\lib\STM32duino STM32SD_ID1810\src/bsp_sd.h:122:26: error: unknown type name 'SD_HandleTypeDef'; did you mean 'SPI_HandleTypeDef'?
>  void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);
>                           ^~~~~~~~~~~~~~~~
>                           SPI_HandleTypeDef
> In file included from C:\Users\fmukhtarif\.platformio\lib\STM32duino STM32SD_ID1810\src/ffconf.h:9:0,
>                  from C:\Users\fmukhtarif\.platformio\lib\FatFs_ID1805\src/ff.h:29,
>                  from C:\Users\fmukhtarif\.platformio\lib\FatFs_ID1805\src/ff_gen_drv.h:58,
>                  from C:\Users\fmukhtarif\.platformio\lib\FatFs_ID1805\src\drivers\sd_diskio.c:49:
> C:\Users\fmukhtarif\.platformio\lib\STM32duino STM32SD_ID1810\src/bsp_sd.h:58:40: error: unknown type name 'HAL_SD_CardInfoTypeDef'; did you mean 'HAL_SD_CardInfoTypedef'?
>  #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef
>                                         ^
> C:\Users\fmukhtarif\.platformio\lib\STM32duino STM32SD_ID1810\src/bsp_sd.h:115:28: note: in expansion of macro 'HAL_SD_CardInfoTypedef'
>  void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);
>                             ^~~~~~~~~~~~~~~~~~~~~~
[... 214 lines cut ...]
> C:\Users\fmukhtarif\.platformio\lib\STM32duino STM32SD_ID1810\src/bsp_sd.h:122:49: error: expected primary-expression before 'void'
>  void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);
>                                                  ^~~~
> C:\Users\fmukhtarif\.platformio\lib\STM32duino STM32SD_ID1810\src/bsp_sd.h:58:40: error: 'HAL_SD_CardInfoTypeDef' does not name a type; did you mean 'HAL_SD_CardInfoTypedef'?
>  #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef
>                                         ^
> C:\Users\fmukhtarif\.platformio\lib\STM32duino STM32SD_ID1810\src/bsp_sd.h:64:21: note: in expansion of macro 'HAL_SD_CardInfoTypedef'
>  #define SD_CardInfo HAL_SD_CardInfoTypedef
>                      ^~~~~~~~~~~~~~~~~~~~~~
> C:\Users\fmukhtarif\.platformio\lib\STM32duino STM32SD_ID1810\src/Sd2Card.h:62:5: note: in expansion of macro 'SD_CardInfo'
>      SD_CardInfo _SdCardInfo;
>      ^~~~~~~~~~~
> src\main.cpp: In function 'void setup()':
> src\main.cpp:73:5: error: 'myFile' was not declared in this scope
>      myFile = SD.open("example.txt", FILE_WRITE);
>      ^~~~~~
> src\main.cpp:73:5: note: suggested alternative: 'File'
>      myFile = SD.open("example.txt", FILE_WRITE);
>      ^~~~~~
>      File
> *** [.pio\build\bluepill_f103c8\src\main.cpp.o] Error 1
> ====================================================== [FAILED] Took 22.20 seconds ======================================================
> The terminal process terminated with exit code: 1
> 
> Terminal will be reused by tasks, press any key to close it.

fpistm, 2020-03-21:
Hi @3nd0y 
This is normal it does not build as BP has no SDIO peripheral.
I guess you try to use an SD card shield over SPI, in this case use the default SD library from Arduino.

closed 2020-03-21 by fpistm

saloid, 2020-07-19:
If your board really have SDIO and your MCU support it, you should specify build flag in platformio: `build_flags = -DHAL_SD_MODULE_ENABLED`. For example, `genericSTM32F103RC` has SDIO, but it is disabled by default