stm32duino/STM32SD #22: Problem reading binary data

GH15ADF, 2020-04-19:
I'm having a problem getting existing code that works on ESP32 and M4 to work on the Adafruit STM32F405 Feather using STM32duino and STM32SD. I believe I have traced this down to a different behavior in the File::read() function behavior between the platforms. 

The code attempts to read a binary plugin file from the SD card. The code assumes that a character (byte) at a time is read and writes the character to a VS1053 DSP connected via SPI. The code understand the structure of the plugin file and calculates the number of bytes to read and load into specific memory locations in the VS1053. The initial symptom was that the number of bytes to read from the plugin file was being computed incorrectly on the STM32.

To reproduce the issue, I created the following sketch that outputs the same symptom I am seeing tracing in the Adafruit code. The expected Serial.print() output is two hex characters, but there are occasional four hex character outputs indicating that the read returned the uint16_t assumed to be only a byte but padded with 1's rather than zero's.  I have determined that any of the *.img files and various other non-ASCII files exhibit the same behavior. Files that are only ASCII text do not seem to exhibit the problem, all outputted values are two hex characters.

```
#include <Arduino.h>
#include <STM32SD.h>

void setup() {
    uint16_t data;
    // uint8_t data;

    Serial.begin(115200);
    while (!Serial)
        ;
    Serial.print("Initializing SD card...");
    while (!SD.begin(SD_DETECT_PIN)) {
        delay(10);
    }
    Serial.println("SD initialization done.");
    File plugin = SD.open("/venc44k1q05.img");

    for (int i = 0; i < 200; i++) {
        data = plugin.read();
        Serial.printf(" %X", data);
    }

    plugin.close();
}

void loop() {
    // put your main code here, to run repeatedly:
}
```

Example out from venc44k1q05.img:

```
Initializing SD card...SD initialization done.
 50 26 48 1 0 20 18 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 18 0 20 77 0 0 1 0 0 0 0 0 0 0 2 0 39 FFBC 0 34 0 0 13 50 FFF4 0 54 4 0 0 8 11 FFF4 0 56 44 FFF4 0 54 4 FFF4 0 56 48 FFF4 0 54 4 FFF4 0 56 54 FFF4 0 54 4 FFF4 0 56 40 0 0 FF80 A 0 C 0 0 0 6 FFC0 10 FFB1 FF82 FFA3 FFC0 0 30 2 51 39 0 60 0 0 30 6 FF91 39 0 40 24 0 0 0 FF90 0 0 40 0 38 10 0 24 0 0 42 0 38 0 0 24 29 0 
62 0 0 6 20 16 2A 0 14 4E 2A 0 1A FFCE 2A 0 28 E 2A 0 45 4E 30 9 38 51 3E 14 FFF8 12 3E 11 FFB8 17 0 0 0 FFD7 0 20 3 FFD2 37 4 FFC0 24 0 30 5 51 3E 11 38 14 3E 2 FF80 24 0 30 3
```

Specifically, the file I am attempting to read is venc44k1q05.img which is a plugin from the "profiles" folder in a download provided by the manufacturer (VLSI) for the VS1053 chip. The link to that download is http://www.vlsi.fi/fileadmin/software/VS10XX/vs1053-vorbis-encoder-170c.zip

Some additional details:

The Adafruit code is https://github.com/adafruit/Adafruit_VS1053_Library/blob/master/Adafruit_VS1053.cpp at line 374, the Adafruit_VS1053::loadPlugin function.

The console output from PlatformIO in VSCode with build details is below:
```
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/adafruit_feather_f405.html
PLATFORM: ST STM32 6.1.0 > Adafruit Feather STM32F405
HARDWARE: STM32F405RGT6 168MHz, 128KB RAM, 1MB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.10800.200207 (1.8.0)
 - tool-dfuutil 1.9.200310 
 - tool-openocd 2.1000.190707 (10.0)
 - tool-stm32duino 1.0.2
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 13 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <STM32duino STM32SD> 1.2.2
|   |-- <FatFs> 2.0.3
```

fpistm, 2020-04-20:
Hi @GH15ADF 
Currenlty, I will not have time to dig into this, I will do when I can.
Anyway, FYI, I already tested to read an image on the SD over SDIO and sent it over SPI (TFT example from Arduino) and it worked.

GH15ADF, 2020-04-20:
Thanks for taking the time to look at this. 

I did look through TFTBitmapLogo.ino and that used TPImage::loadImage (from Adafruit_GFX.h) which ultimately casts every read as a byte (I believe). I added a cast to byte to the example sketch and the file read works as expected, so I have a work around.

I also looked at the original code from VLSI (vs1053oggrec.c in the vs1053-vorbis-encoder-170c.zip) and it used the standard C language fgetc() function. According to the C99 spec:

_...the fgetc function obtains that character as an unsigned char converted to an int..._

So I believe the upper byte of the int returned not being 0 is unexpected and the Arduino port of vs1053oggrec.c handles read() like fgetc().

Anyway, I thought I would point this out in case it turns out to be a real bug.

fpistm, 2020-08-13:
Hi @GH15ADF 
you're right all are cast as a byte and you point the right thing I guess.
The read API return an `int `to be able to manage the error `-1` anyway it read only a byte.
https://www.arduino.cc/en/Reference/FileRead

I guess the library code could be hardened as no check is done on the read value:
https://github.com/adafruit/Adafruit_VS1053_Library/blob/379f335da90d8fe287b6fe423c14551e55f87c8e/Adafruit_VS1053.cpp#L410-L432

fpistm, 2020-09-15:
Close as library should properly check the read value. API is inline with the Arduino reference and have to be used properly as done in the `TPImage::loadImage (from Adafruit_GFX.h) `

closed 2020-09-15 by fpistm