stm32duino/STM32SD #23: STM32F407VGT can not compile

STAFFD, 2020-04-20:
I am currently using this library with STM32F407VGT which contains SDIO. But I still fail when compilling. I read those closed issues for solution and noticed that this problem can occur when using those STM32 MCUs without  SDIO. So I enabled HAL_SD_MODULE_ENABLED added the PeripheralPins.h before including STM32SD.h.  But I still get an error.

`/Users/sheldonvon/Library/Arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: libraries/STM32SD/bsp_sd.c.o: in function `BSP_SD_MspInit':
bsp_sd.c:(.text.BSP_SD_MspInit+0x34): undefined reference to `PinMap_SD'
collect2: error: ld returned 1 exit status
exit status 1`

Can you please tell me why this happened and how to solve this?

fpistm, 2020-04-20:
> So I enabled HAL_SD_MODULE_ENABLED added the PeripheralPins.h before including STM32SD.h.

How did you do that ?
Which variant you used and core version ?

STAFFD, 2020-04-20:
> > So I enabled HAL_SD_MODULE_ENABLED added the PeripheralPins.h before including STM32SD.h.
> 
> How did you do that ?
> Which variant you used and core version ?

I used DIYMORE_F407VGT for board configuration. Indeed, I failed to find anything related to SDIO under this folder ` ~/Library/Arduino15/packages/STM32/hardware/stm32/1.8.0/variants/DIYMORE_F407VGT`, but the datasheet of STM32F407VGT did mention that it contains SDIO. 

For enabling HAL_SD_MODULE, I simply added `#define HAL_SD_MODULE_ENABLED` in the file stm32yyxx_hal_config.h. 

And I also added PeripheralPins.h like this:

`
#include <PeripheralPins.h>
#include <STM32SD.h>
`
I am not professional in this area, so please tell me if I did anything wrong.

fpistm, 2020-04-20:
Well,

you don't have to modify the core files.

To enable the HAL SD module, simply add a file at sketch level: `hal_conf_extra.h`
Then in this file add:
`#define HAL_SD_MODULE_ENABLED `

For the pin map issue, then you're right the PeripheralPins.c currently does not contains the SD pin map array.
You don't have to include it in your sketch, you have to define this array, you can define it in your sketch, the full array is available here:
https://github.com/stm32duino/Arduino_Tools/blob/master/src/genpinmap/Arduino/STM32F4/STM32F407V(E-G)Tx/PeripheralPins.c#L379-L395

```C
const PinMap PinMap_SD[] = {
  //  {PB_8,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D4
  //  {PB_9,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D5
  //  {PC_6,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D6
  //  {PC_7,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D7
  {PC_8,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D0
  {PC_9,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D1
  {PC_10, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D2
  {PC_11, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D3
  {PC_12, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, GPIO_AF12_SDIO)}, // SDIO_CK
  {PD_2,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, GPIO_AF12_SDIO)}, // SDIO_CMD
  {NC,    NP,    0}
};
```

STAFFD, 2020-04-20:
> I guess you wired yourself a SD card slot as the DIYMORE F407 does not have by default a SD card slot ?
> https://stm32-base.org/boards/STM32F407VGT6-diymore

Well, my board is not exactly DIYMORE F407. It has a SD card slot by defaul. I use DIYMORE F407 because it is using the same STM32F4 with my board. So the wiring must be correct.

STAFFD, 2020-04-20:
> Well,
>
> you don't have to modify the core files.
>
> To enable the HAL SD module, simply add a file at sketch level: `hal_conf_extra.h`
> Then in this file add:
> `#define HAL_SD_MODULE_ENABLED `
>
> For the [...]

Thanks, the first solution is helpful, and as for the PinMap_SD, I made a little change at this function

`
__weak void BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params)
{
  UNUSED(Params);

  /* Configure SD GPIOs */
  const PinMap *map = {
    {PC_8,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D0
    {PC_9,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D1
    {PC_10, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D2
    {PC_11, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D3
    {PC_12, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, GPIO_AF12_SDIO)}, // SDIO_CK
    {PD_2,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, GPIO_AF12_SDIO)}, // SDIO_CMD
    {NC,    NP,    0}
  };
  while (map->pin != NC) {
    pin_function(map->pin, map->function);
    map++;
  }
`
The compilation was successful, however the initialization of SD card failed.

fpistm, 2020-04-20:
This is not a function, this is only an array, so you don't have to  wrote a BSP_SD_MspInit().
The GPIO will be initialized properly with what I provided.
If it does not init prperly maybe you have a SD card detect pin ?
In that case define it in the hal_conf_extra.h (https://github.com/stm32duino/STM32SD#sd-detect-and-timeout)

`#define SD_DETECT_PIN`

STAFFD, 2020-04-20:
> This is not a function, this is only an array, so you don't have to wrote a BSP_SD_MspInit().
> The GPIO will be initialized properly with what I provided.
> If it does not init prperly maybe you have a SD card detect pin ?
> In that case define it in the hal_conf_extra.h (https://github.com/stm32duino/STM32SD#sd-detect-and-timeout)
> 
> `#define SD_DETECT_PIN PYn`

I meant I made a modification in 
`__weak void BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params)`
and  I dont think this is an array.

I believe I dont have a detect pin. Here is the circuit of my board.
[image]

By the way, it is possible that there are some bugs in DIYMORE F407. Because I got an error telling me Serial3 was an undefined symbol when I was using Serial3 which is obviously available.

fpistm, 2020-04-20:
Well,I know it is a function :wink: I talk about the PinMap array...  please do not modify the BSP_SD_MspInit function.
Simply declare at sketch level the PinMap as I told you.
Because by changing the MSP init function it seems you removed the SD CLK enable...

fpistm, 2020-04-20:
> By the way, it is possible that there are some bugs in DIYMORE F407. Because I got an error telling me Serial3 was an undefined symbol when I was using Serial3 which is obviously available.

No, Serial3 is not available by default for the DIYMORE F407... the default is Serial1 which is mapped on Serial generic instance name: https://github.com/stm32duino/Arduino_Core_STM32/blob/6ef6b0d2d99dcb39175c4a63f411aceb9900354a/variants/DIYMORE_F407VGT/variant.h#L170
If you want Serial3 you have to instantiate it, see https://github.com/stm32duino/wiki/wiki/API#hardwareserial
This library have been tested on F407 thanks the Black F407 board which include by default a SD card slot. 
There is no build issue if you define properly you board.
For this kind of support please use the forum instead: https://www.stm32duino.com

closed 2020-04-20 by fpistm

STAFFD, 2020-04-20:
> No, Serial3 is not available by default for the DIYMORE F407... the default is Serial1 which is mapped on Serial generic instance name: [...]

Oh, I see! I run more test on my board! Thanks a lot !!