stm32duino/STM32SD #26: SDIO Clock Configuration

CMGeldenhuys, 2020-07-28:
I'm trying to get the SD module working on a Nucleo-F446RE. This board does not have an SD card slot and thus by default does not have a pin mapping for the SD module. I've created the following pin mapping along with the `-DHAL_SD_MODULE_ENABLED` build flag.
``` cpp
const PinMap PinMap_SD[] = {
  {PC_8,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D0
  {PB_0,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D1
  {PB_1, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)},  // SDIO_D2
  {PC_11, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D3
  {PB_2, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, GPIO_AF12_SDIO)},  // SDIO_CK
  {PD_2,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, GPIO_AF12_SDIO)}, // SDIO_CMD
  {NC,    NP,    0}
};
```
Running both the _CardInfo_ and _listfiles_ examples lead to either an infinite loop or _"initialization failed. Is a card inserted?"_ error being printed to the serial. Since the SD card module is on a breakout board with long jumper leads I suspected it might be a clock issue. But I found no documented way of changing the clock speed divider (which is present in CubeMX). 

However, manually changing [line 141](https://github.com/stm32duino/STM32SD/blob/68eeafa4eadf2e838a36d3f4779b3c30534fc779/src/bsp_sd.c#L141) seemed to do the trick and got everything working. 
https://github.com/stm32duino/STM32SD/blob/68eeafa4eadf2e838a36d3f4779b3c30534fc779/src/bsp_sd.c#L138-L141

Is there a better way to change the `SD_CLK_DIV`? Or is this option configured elsewhere? I did try _override_ the option in the user sketch using `#undef SD_CLK_DIV` and then `#define SD_CLK_DIV 4`, but this didn't seem to work. Maybe because the `SD_CLK_DIV` is declared inside of the `bsp_sd.c` file and not the `bsp_sd.h` file?

fpistm, 2020-08-13:
Hi @CMGeldenhuys 
As by default the Nucleo F446RE does not have a SD card slot, the clock is not configured.
I think you should try to configure it, like this you will not need to change the `SD_CLK_DIV`.

You could redefine the `WEAK void SystemClock_Config(void)` at sketch level:

```C++
extern "C" void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {};
  RCC_PeriphCLKInitTypeDef PeriphClkInitStruct = {};

  /* Configure the main internal regulator output voltage  */
  __HAL_RCC_PWR_CLK_ENABLE();
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);
  /*
   * Initializes the RCC Oscillators according to the specified parameters
   * in the RCC_OscInitTypeDef structure.
   */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSI;
  RCC_OscInitStruct.PLL.PLLM = 8;
  RCC_OscInitStruct.PLL.PLLN = 180;
  RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
  RCC_OscInitStruct.PLL.PLLQ = 8;
  RCC_OscInitStruct.PLL.PLLR = 2;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK) {
    Error_Handler();
  }
  /* Activate the Over-Drive mode */
  if (HAL_PWREx_EnableOverDrive() != HAL_OK) {
    Error_Handler();
  }
  /* Initializes the CPU, AHB and APB buses clocks */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
                              |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV4;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV2;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_5) != HAL_OK) {
    Error_Handler();
  }
  PeriphClkInitStruct.PeriphClockSelection = RCC_PERIPHCLK_SDIO|RCC_PERIPHCLK_CLK48;
  PeriphClkInitStruct.Clk48ClockSelection = RCC_CLK48CLKSOURCE_PLLQ;
  PeriphClkInitStruct.SdioClockSelection = RCC_SDIOCLKSOURCE_CLK48;
  if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInitStruct) != HAL_OK) {
    Error_Handler();
  }
}
```

With this config the SD clock input will be 45MHz instead of 50MHz which is out of range, clock have to be 48MHz < SD clock < 25 MHz.

If this is OK I  will change the default clock config and also add the SD pinMap in peripheralPins.c.

fpistm, 2020-10-20:
Closed as no feedback from OP.
PeripheralPins.c will be updated soon after all variant rework done in the core. For clock config, user can override it.

closed 2020-10-20 by fpistm