stm32duino/STM32SD #27: This library NOT suitable for Arduino IDE using SPI connection

LimCheeKheong, 2020-09-02:
Compiling a basic program "CardInfo" resukted in the following errors (just partial attached) as:-

In file included from C:\Users\xxx\Documents\Arduino\libraries\STM32SD-master\src/Sd2Card.h:40,
                 from C:\Users\xxx\Documents\Arduino\libraries\STM32SD-master\src/STM32SD.h:22,
                 from C:\Users\xxx\Documents\Arduino\STM32F4_4Com\STM32F4_SD_TFT_RW1\STM32F4_SD_TFT_RW1.ino:21:
C:\Users\xxx\Documents\Arduino\libraries\STM32SD-master\src/bsp_sd.h:61:40: error: variable or field 'BSP_SD_GetCardInfo' declared void
   61 | #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef
      |                                        ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\xxx\Documents\Arduino\libraries\STM32SD-master\src/bsp_sd.h:118:28: note: in expansion of macro 'HAL_SD_CardInfoTypedef'
  118 | void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);
      |                            ^~~~~~~~~~~~~~~~~~~~~~

Several authors mentioned about define HAL_SD_MODULE_ENABLED but I am not sure define it on which line or where.  I simply put it here as:-
/*SD Card information structure */
#ifndef STM32L1xx
#define HAL_SD_MODULE_ENABLED
#define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef
#define BSP_SD_CardInfo                HAL_SD_CardInfoTypeDef
#define HAL_SD_WideBusOperation_Config HAL_SD_ConfigWideBusOperation
#define HAL_SD_Get_CardInfo            HAL_SD_GetCardInfo
#endif

It still doesn't work.

I also saw fpistm mentioned this "After reading again the thread, I guess several user try to use this library for SD card connected through SPI while it is for SD Card connected through SDIO interface."

So this library is NOT suitable SPI connection .

fpistm, 2020-09-02:
Yes this library is not for SD through SPI. If you want use SPI simply use the default SD Arduino library.
So what is your issue? And what do you want to do?
Which board? Which SD connection? ....

LimCheeKheong, 2020-09-02:
I am using WeAct STM32F401CE board (however as this board is not listed, I am using STM32F401CC). When compiling using normal Arduino SD - here is a warning notice from compiler

WARNING: library SPI claims to run on stm32 architecture(s) and may be incompatible with your current board which runs on STM32 architecture(s).
WARNING: library SrcWrapper claims to run on stm32 architecture(s) and may be incompatible with your current board which runs on STM32 architecture(s).
WARNING: library Wire claims to run on stm32 architecture(s) and may be incompatible with your current board which runs on STM32 architecture(s).

It still runs ok using Arduino SD.h   even with warning.

closed 2020-09-02 by LimCheeKheong

fpistm, 2020-09-02:
Well, as the SPI library is a builtin on it is not possible that it claims to run on stm32 architecture as the core core is for stm32.
I don't know what is your environment but it seems you have an issue.