stm32duino/STM32SD #28: Bad formating?

mrguen, 2020-09-14:
Hello,

This library seems great! I tried the CardInfo.ino example. But it can't list files. I used a brand new SDHC card an windows sees that it is formated in FAT32.

Here is the visual monitor output:

`Initializing SD card...A card is present.

Card type: SDHC

Volume type is FAT32

Volume size (bytes): 15971909632
Volume size (Kbytes): 15597568
Volume size (Mbytes): 15232

Files found on the card (name, date and size in bytes): 
System Volume Information
  WPSettings.dat 2020-09-14 19:33:18 12
  IndexerVolumeGuid 2020-09-14 19:33:18 76
⸮⸮⸮⸮⸮⸮⸮⸮.⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮.⸮⸮⸮
Error to open dir: ⸮⸮⸮⸮⸮⸮⸮⸮.⸮⸮⸮
⸮⸮⸮⸮⸮⸮⸮⸮.⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮.⸮⸮⸮
Error to open dir: ⸮⸮⸮⸮⸮⸮⸮⸮.⸮⸮⸮
`

I am in France, using a french system... I suppose there is a configuration issue. Could you give me your advice?

Thanks

fpistm, 2020-09-15:
Hi @mrguen 
Which kind of file it is ? Names ?
Could you share it. My first guess is it could be linked to LFN or format 8.3.

mrguen, 2020-09-15:
Hello @fpistm  

ok, in fact windows can't see it as an SD card. I could acces it through a USB adapter. It is seen as a USB drive in FAT32 format.  I don't know how to recognize / format it otherwise.

fpistm, 2020-09-15:
For Windows, it seems correct an SD card is simply a mass storage so see it as an USB drive is correct, I guess.

mrguen, 2020-09-15:
- ok, I had to format it again under windows. 

Now

- Example Cardinfo.ino works to list a file created under windows

- ReadWrite.ino:

"
Initializing SD card...initialization done.
Writing to test.txt...done.
error opening test.txt
###### End of the SD tests ######
"
In fact the file was not written (nothing visble under windows)

- Another possibly strange behaviour: either when setting SD_DETECT_PIN SD_DETECT_NONE or SD_DETECT_PIN PC4 the device is not found after downloading. I have to pushed it out and into the slot. After this it is seen immediatly when restting the device and opening the monitor again.

fpistm, 2020-09-15:
Where did you set `SD_DETECT_PIN`?

mrguen, 2020-09-15:
I tried 

- without changing anything i.e. defined in the sketch 

- and changing it in the sketch

fpistm, 2020-09-15:
OK Fine.
Which variant you used ? The generic F401Rx ?

mrguen, 2020-09-15:
Yes

fpistm, 2020-09-15:
OK Thanks for the quick feedback.
Those variants has not take in account the SD part.
So, I guess you have enabled the `HAL_SD_MODULE_DISABLED` else it won't compile.
By chance there is no duplicated pin in the `PinMap_SD` array so the good pins are initialized.
But what I suspect is the clock config is not correct which could explain those issues. The clock have to be less or equal to 48 MHz IIWR.

Anyway, I will have to rework this library to ease use of SD pins as I've done to the [MX25R6435F](https://github.com/stm32duino/MX25R6435F) library.

mrguen, 2020-09-15:
Ok; Thanks. Strange the behaviour seems unconstant indeed. 
Yes, I used -DHAL_SD_MODULE_ENABLED in build_opt.h

fpistm, 2020-09-15:
On your custom board do you have an HSE ? If yes which value ?

mrguen, 2020-09-15:
8 MHz

fpistm, 2020-09-15:
OK. The clock config seems ok for SD.
Thanks to the schematics, it seems other SD pins are connected and as I said all of them are initialized maybe this could cause some issue.
Could you try to redefine the `PinMap_SD `with only the used pins on your sketch:

```
extern "C" const PinMap PinMap_SD[] = {
  {PC_8,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D0
  {PC_9,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D1
  {PC_10, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D2
  {PC_11, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D3
  {PC_12, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, GPIO_AF12_SDIO)}, // SDIO_CK
  {PD_2,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, GPIO_AF12_SDIO)}, // SDIO_CMD
  {NC,    NP,    0}
};

```

mrguen, 2020-09-15:
"conflicting declaration of 'const PinMap PinMap_SD []' with 'C' linkage"

fpistm, 2020-09-15:
OK. At this time I've no more clue for you, sorry. I guess your SD is not write protected?

mrguen, 2020-09-15:
No. The behaviour is strange. I just wrote a  "essai.txt" file under windows. Then used ReadWrite.ino: This time the test.txt file was created but the older essai.txt file was removed. Sometimes the file is written, sometimes not and opening it is faulty every time. Also the card detect thing is not constant. I thought about a hardware / physical plugin problem but with the other symptoms it looks like a software / file system problem.

fpistm, 2020-09-15:
Well hard to tell. You best bet is to debug.

mrguen, 2020-09-15:
to debug... down the libraries with Serial output?

fpistm, 2020-09-15:
No, I mean use a STLink, a GDB,...
I could not help more on this subject as I don't have your hardware. I don't tell there is no software issue but I could not investigate on this without be able to reproduce.
Maybe the next FatFS update will solve this issue but currently I have not time to update it.

mrguen, 2020-09-22:
@fpistm 

- From reading https://github.com/stm32duino/STM32SD#configuration I understand that we can change some configuration parameters.  Since it includes a couple of encoding parameters, I suppose it might help to solve the problem explained above.

- I did not find a file ffconf.h in C:\Users\TG\Documents\Arduino\libraries\FatFs\src so I copied ffconf_template.h to ffconf.h

- But now I get the error "C:\Users\TG\Documents\Arduino\libraries\FatFs\src/ffconf.h:290:10: fatal error: cmsis_os.h: No such file or directory"

fpistm, 2020-09-22:
As it is explained in the README.md:
https://github.com/stm32duino/STM32SD#dependency

This library depends on FatFs library where you will find the ffconf.h...

mrguen, 2020-09-22:
@fpistm Yes, I am speaking about the FaFs library ffconf.h in C:\Users\TG\Documents\Arduino\libraries\FatFs\src  this is the library linked since the compiler reported the bug. 
[image]

fpistm, 2020-09-22:
> User can provide his own defined options by adding his configuration in a file named ffconf_custom.h at sketch level or in variant folder.

fpistm, 2020-09-22:
If you modified the Library structure then don't be surprised to have a build issue. This is not a bug as this is your modification which cause the bug.

mrguen, 2020-09-22:
1) I did not modify anything. Just used the ffconf_template.h as ffconf.h as explained in https://github.com/stm32duino/STM32SD#configuration 
"The FatFs has several user defined options, which is specified from within the ffconf.h file.
This library provides a default user defined options file named ffconf_default.h"

I did not find a ffconf_default.h but a ffconf_template.h so I used it. 

2) I just tried without any change in the FaFs library (removed the new ffconf.h) and adding the ffconf_default.h as ffconf_custom.h at sketch level and get the same error

"C:\Users\TG\Documents\Arduino\STM32\CardInfo/ffconf_custom.h:291:10: fatal error: cmsis_os.h: No such file or directory"

fpistm, 2020-09-22:
This is not what is explained.

> The FatFs has several user defined options, which is specified from within the ffconf.h file.

This only explained the FatFs use this file not you have to change it.

> This library provides a default user defined options file named ffconf_default.h.

This line explained a default file is provided in this library.

> User can provide his own defined options by adding his configuration in a file named ffconf_custom.h at sketch level or in variant folder.

And now this line explained the way to redefine your own options.

If you open the `ff_conf.h` you would see how it is managed:

https://github.com/stm32duino/STM32SD/blob/6f7c3326f51cef14fa2411c1b0fb8ba28ed139ff/src/ffconf.h#L11-L21

mrguen, 2020-09-22:
Ok, we are not speaking about the same file. I was refering to the library conf file in C:\Users\TG\Documents\Arduino\libraries\FatFs\src

and you are refering to STM32SD/src/ffconf.h which has no other option than _FATFS while the original file has a couple of encoding options.

mrguen, 2020-09-22:
Ok, I will try to use ffconf_default_68300.h as ffconf_custom.h and tweak the encoding parameters to use unicode.

mrguen, 2020-09-22:
It does compile when using ffconf_default_68300.h as ffconf_custom.h  at sketch level.

When changing the option "_LFN_UNICODE  0" to "_LFN_UNICODE  1", a couple of bugs.

First one:
"
C:\Users\TG\Documents\Arduino\libraries\STM32duino_STM32SD\src\SD.cpp: In static member function 'static bool SDClass::exists(const char*)':
C:\Users\TG\Documents\Arduino\libraries\STM32duino_STM32SD\src\SD.cpp:83:14: error: cannot convert 'const char*' to 'const TCHAR*' {aka 'const short unsigned int*'}
   83 |   if (f_stat(filepath, &fno) != FR_OK) {
      |              ^~~~~~~~
      |              |
      |              const char*
In file included from C:\Users\TG\Documents\Arduino\libraries\FatFs\src/ff_gen_drv.h:58,
                 from C:\Users\TG\Documents\Arduino\libraries\FatFs\src/FatFs.h:42,
                 from C:\Users\TG\Documents\Arduino\libraries\STM32duino_STM32SD\src\SdFatFs.h:43,
                 from C:\Users\TG\Documents\Arduino\libraries\STM32duino_STM32SD\src\STM32SD.h:23,
                 from C:\Users\TG\Documents\Arduino\libraries\STM32duino_STM32SD\src\SD.cpp:56:
C:\Users\TG\Documents\Arduino\libraries\FatFs\src/ff.h:257:30: note:   initializing argument 1 of 'FRESULT f_stat(const TCHAR*, FILINFO*)'
  257 | FRESULT f_stat (const TCHAR* path, FILINFO* fno);     /* Get file status */

"

fpistm, 2020-09-22:
This is not a surprise. Not all config has been tested, goal of this library was to be in line with default SD library IIRW. So LFN was not supported.

mrguen, 2020-09-22:
Ok. Well, I am not competent to debug this. The only thing I can say is that it does not work on my configuration. I believe the hardware layer is ok since CardInfo reports the type of SD card, and sizes but it can't list the files properly, printing only  ????? characters. Probably I will wait for the rework.

fpistm, 2020-09-22:
yes propbably printing. anyway this is the first time I see that.
Could you share one of the file for reference ?

mrguen, 2020-09-22:
which file?

fpistm, 2020-09-22:
> 
> 
> which file?

One of the sdcard files which is not properly displayed.

mrguen, 2020-09-22:
I tried a sequence of writing to the file, using CardInfo.ino and ReadWrite.ino

I could not get the ???? characters error when trying to list the content. But I still get these problems

- card not detected when launching sketches. Need to push it back again in the slot
- corrupted file system reported by windows  once
- ReadWrite.ino reports it wrote to the file but can't open it. Does not write content into the file.

- Output of CardInfo.ino on a freshly formated SDHC card level 10 in FAT 32 4096 bytes 
```
A card is present.

Card type: SDHC

Volume type is FAT32

Volume size (bytes): 15942549504
Volume size (Kbytes): 15568896
Volume size (Mbytes): 15204

Files found on the card (name, date and size in bytes): 
System Volume Information
  WPSettings.dat 2020-09-22 12:01:36 12
  IndexerVolumeGuid 2020-09-22 12:01:40 76
###### End of the SD tests ######

```

- ReadWrite.ino:

```
Initializing SD card...initialization done.
Writing to test.txt...done.
**error opening test.txt**
###### End of the SD tests ######
```

- CardInfo.ino again
**Same as first time. Does not see any file**

- **Trying to create a file under windows gives an error 0x80070570 corrupted repertory**

- Formating the SD card again under win 10 (fr-fr) and creating an empty file "test.txt"

-Using CardInfo.ino again

"
A card is present.

Card type: SDHC

Volume type is FAT32

Volume size (bytes): 15942549504
Volume size (Kbytes): 15568896
Volume size (Mbytes): 15204

Files found on the card (name, date and size in bytes): 
System Volume Information
  WPSettings.dat 2020-09-22 12:17:20 12
  IndexerVolumeGuid 2020-09-22 12:17:24 76
Test.txt 2020-09-22 12:17:28 0
###### End of the SD tests ######
"

- ReadWrite.ino
"
Initializing SD card...initialization done.
Writing to test.txt...done.
**error opening test.txt**
###### End of the SD tests ######
"

- CardInfo.ino

"
A card is present.

Card type: SDHC

Volume type is FAT32

Volume size (bytes): 15942549504
Volume size (Kbytes): 15568896
Volume size (Mbytes): 15204

Files found on the card (name, date and size in bytes): 
System Volume Information
  WPSettings.dat 2020-09-22 12:17:20 12
  IndexerVolumeGuid 2020-09-22 12:17:24 76
Test.txt 2020-09-22 12:17:28 0
###### End of the SD tests ######
"

**Reading the file under windows: empty**
Under windows, adding content "test" into the file

- ReadWrite.ino
"
Initializing SD card...initialization done.
Writing to test.txt...done.
error opening test.txt
###### End of the SD tests ######
"

- **Opening the file under windows: not modified by ReadWrite.ino**

fpistm, 2020-09-22:
Did you try with an other SD card ?

mrguen, 2020-09-23:
Maybe this is the answer: not reliable with build in pullups https://community.st.com/s/question/0D50X00009XkfrVSAR/using-internal-pullups-for-sdio-interface

jvictorsoto, 2021-03-27:
I've found a similar weird issue where SD stops working after a while, sometimes don't even start working... my board has 25Mhz crystal, by default SD_CLK_DIV is defined to 0, chaging it to 4 fixed it. Not sure of which freq can handle the card, but for my transfer rates its ok.

closed 2021-06-02 by fpistm