stm32duino/STM32SD #29: STM32F103VET6 problem

saeed14473, 2020-09-19:
Hi, what should I do for this error? 
`In file included from C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\src/Sd2Card.h:40,
                 from C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\src/STM32SD.h:22,
                 from C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\examples\CardInfo\CardInfo.ino:12:
C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\src/bsp_sd.h:58:40: error: variable or field 'BSP_SD_GetCardInfo' declared void
   58 | #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef
      |                                        ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\src/bsp_sd.h:115:28: note: in expansion of macro 'HAL_SD_CardInfoTypedef'
  115 | void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);
      |                            ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\src/bsp_sd.h:58:40: error: 'HAL_SD_CardInfoTypeDef' was not declared in this scope; did you mean 'HAL_SD_CardInfoTypedef'?
   58 | #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef
      |                                        ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\src/bsp_sd.h:115:28: note: in expansion of macro 'HAL_SD_CardInfoTypedef'
  115 | void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);
      |                            ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\src/bsp_sd.h:115:52: error: 'CardInfo' was not declared in this scope; did you mean 'SD_CardInfo'?
  115 | void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);
      |                                                    ^~~~~~~~
      |                                                    SD_CardInfo
C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\src/bsp_sd.h:120:24: error: variable or field 'BSP_SD_MspInit' declared void
  120 | void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                        ^~~~~~~~~~~~~~~~
C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\src/bsp_sd.h:120:24: error: 'SD_HandleTypeDef' was not declared in this scope; did you mean 'SPI_HandleTypeDef'?
  120 | void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                        ^~~~~~~~~~~~~~~~
      |                        SPI_HandleTypeDef
C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\src/bsp_sd.h:120:42: error: 'hsd' was not declared in this scope
  120 | void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                                          ^~~
C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\src/bsp_sd.h:120:47: error: expected primary-expression before 'void'
  120 | void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                                               ^~~~
C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\src/bsp_sd.h:121:31: error: variable or field 'BSP_SD_Detect_MspInit' declared void
  121 | void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                               ^~~~~~~~~~~~~~~~
C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\src/bsp_sd.h:121:31: error: 'SD_HandleTypeDef' was not declared in this scope; did you mean 'SPI_HandleTypeDef'?
  121 | void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                               ^~~~~~~~~~~~~~~~
      |                               SPI_HandleTypeDef
C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\src/bsp_sd.h:121:49: error: 'hsd' was not declared in this scope
  121 | void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                                                 ^~~
C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\src/bsp_sd.h:121:54: error: expected primary-expression before 'void'
  121 | void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                                                      ^~~~
C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\src/bsp_sd.h:122:26: error: variable or field 'BSP_SD_MspDeInit' declared void
  122 | void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);
      |                          ^~~~~~~~~~~~~~~~
C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\src/bsp_sd.h:122:26: error: 'SD_HandleTypeDef' was not declared in this scope; did you mean 'SPI_HandleTypeDef'?
  122 | void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);
      |                          ^~~~~~~~~~~~~~~~
      |                          SPI_HandleTypeDef
C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\src/bsp_sd.h:122:44: error: 'hsd' was not declared in this scope
  122 | void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);
      |                                            ^~~
C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\src/bsp_sd.h:122:49: error: expected primary-expression before 'void'
  122 | void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);
      |                                                 ^~~~
C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\src/bsp_sd.h:58:40: error: 'HAL_SD_CardInfoTypeDef' does not name a type; did you mean 'HAL_SD_CardInfoTypedef'?
   58 | #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef
      |                                        ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\src/bsp_sd.h:64:21: note: in expansion of macro 'HAL_SD_CardInfoTypedef'
   64 | #define SD_CardInfo HAL_SD_CardInfoTypedef
      |                     ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\lenovo\Documents\Arduino\libraries\STM32duino_STM32SD-1.2.2\src/Sd2Card.h:62:5: note: in expansion of macro 'SD_CardInfo'
   62 |     SD_CardInfo _SdCardInfo;
      |     ^~~~~~~~~~~
`

fpistm, 2020-09-19:
Hi, you have to define `HAL_SD_MODULE_ENABLED`.  See https://github.com/stm32duino/wiki/wiki/HAL-configuration#customize-hal-or-variant-definition

closed 2020-09-21 by fpistm

fpistm, 2020-09-21:
Note that `HAL_SD_MODULE_ENABLED `is already defined for the STM32F103VE target:
https://github.com/stm32duino/Arduino_Core_STM32/blob/2fe15c1c36c05464267e8b4316599fc19864e67f/variants/Generic_F103Vx/variant.h#L146

So, I think you used another core than this one: https://github.com/stm32duino/Arduino_Core_STM32

This library is only for this core.

saeed14473, 2020-09-21:
> 
> 
> Note that `HAL_SD_MODULE_ENABLED `is already defined for the STM32F103VE target:
> https://github.com/stm32duino/Arduino_Core_STM32/blob/2fe15c1c36c05464267e8b4316599fc19864e67f/variants/Generic_F103Vx/variant.h#L146
> 
> So, I think you used another core than this one: https://github.com/stm32duino/Arduino_Core_STM32
> 
> This library is only for this core.
> 
> I've built SD example without any error.

I used the same core!

fpistm, 2020-09-21:
Hi @saeed14473 
OK.
So you used Arduino IDE ? The latest STM32 core v1.9.0? And you used the Generic F103VE ?

saeed14473, 2020-09-21:
Arduino IDE 1.8.13
STM32 core 1.9.0
Generic STM32F1 series
Board part number: STM32F103VET6 STUno

fpistm, 2020-09-21:
I never seen this board part number: STM32F103VET6 STUno

So I guess you have a custom environment as it builds properly with the default one. I can't help if you do not provide all relevant information.

Maybe you use this:
https://github.com/r-pv/STM32F103VET6_STUno

So in that case you have to define the `HAL_SD_MODULE_ENABLED  `as I've mentioned earlier.

saeed14473, 2020-09-21:
Yes, thank you for your responses, I'll try it.