stm32duino/STM32SD #34: Lib not work

SymbiSoft, 2020-10-29:
Why not work STM32SD lib? Default sketch - dataloger. Not compiling for bluepill(f103c8t6). 

`In file included from C:\Users\\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\src/Sd2Card.h:40,
                 from C:\Users\\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\src/STM32SD.h:22,
                 from C:\Users\Игорь\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\examples\Datalogger\Datalogger.ino:13:
C:\Users\\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:58:40: error: variable or field 'BSP_SD_GetCardInfo' declared void
   58 | #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef
      |                                        ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:115:28: note: in expansion of macro 'HAL_SD_CardInfoTypedef'
  115 | void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);
      |                            ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:58:40: error: 'HAL_SD_CardInfoTypeDef' was not declared in this scope; did you mean 'HAL_SD_CardInfoTypedef'?
   58 | #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef
      |                                        ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:115:28: note: in expansion of macro 'HAL_SD_CardInfoTypedef'
  115 | void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);
      |                            ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:115:52: error: 'CardInfo' was not declared in this scope; did you mean 'SD_CardInfo'?
  115 | void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);
      |                                                    ^~~~~~~~
      |                                                    SD_CardInfo
C:\Users\\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:120:24: error: variable or field 'BSP_SD_MspInit' declared void
  120 | void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                        ^~~~~~~~~~~~~~~~
C:\Users\\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:120:24: error: 'SD_HandleTypeDef' was not declared in this scope; did you mean 'SPI_HandleTypeDef'?
  120 | void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                        ^~~~~~~~~~~~~~~~
      |                        SPI_HandleTypeDef
C:\Users\\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:120:42: error: 'hsd' was not declared in this scope
  120 | void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                                          ^~~
C:\Users\\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:120:47: error: expected primary-expression before 'void'
  120 | void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                                               ^~~~
C:\Users\\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:121:31: error: variable or field 'BSP_SD_Detect_MspInit' declared void
  121 | void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                               ^~~~~~~~~~~~~~~~
C:\Users\\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:121:31: error: 'SD_HandleTypeDef' was not declared in this scope; did you mean 'SPI_HandleTypeDef'?
  121 | void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                               ^~~~~~~~~~~~~~~~
      |                               SPI_HandleTypeDef
C:\Users\\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:121:49: error: 'hsd' was not declared in this scope
  121 | void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                                                 ^~~
C:\Users\\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:121:54: error: expected primary-expression before 'void'
  121 | void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                                                      ^~~~
C:\Users\\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:122:26: error: variable or field 'BSP_SD_MspDeInit' declared void
  122 | void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);
      |                          ^~~~~~~~~~~~~~~~
C:\Users\\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:122:26: error: 'SD_HandleTypeDef' was not declared in this scope; did you mean 'SPI_HandleTypeDef'?
  122 | void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);
      |                          ^~~~~~~~~~~~~~~~
      |                          SPI_HandleTypeDef
C:\Users\\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:122:44: error: 'hsd' was not declared in this scope
  122 | void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);
      |                                            ^~~
C:\Users\\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:122:49: error: expected primary-expression before 'void'
  122 | void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);
      |                                                 ^~~~
C:\Users\\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:58:40: error: 'HAL_SD_CardInfoTypeDef' does not name a type; did you mean 'HAL_SD_CardInfoTypedef'?
   58 | #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef
      |                                        ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:64:21: note: in expansion of macro 'HAL_SD_CardInfoTypedef'
   64 | #define SD_CardInfo HAL_SD_CardInfoTypedef
      |                     ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\\OneDrive\Documents\Arduino\libraries\STM32duino_STM32SD\src/Sd2Card.h:62:5: note: in expansion of macro 'SD_CardInfo'
   62 |     SD_CardInfo _SdCardInfo;
      |     ^~~~~~~~~~~
exit status 1
Error compiling for board Generic STM32F1 series.
`

AnHardt, 2020-10-30:
Blue Pills CPU (STM32F103C8T6) does noes have SDIO/MMC hardware.
Probably your SD-slot is connected via SPI. In that case use Arduinos build in SD-library (SD.h).

fpistm, 2020-10-30:
Hi @SymbiSoft,
@AnHardt is right you can not use this library with STM32F103C8.

closed 2020-10-30 by fpistm