stm32duino/STM32SD #35: Does not compile for STM32F1

ShaggyDog18, 2021-01-04:
Got compilation errors related to the library. You can recreate the issue by compiling an example Datalogger (I get the same set of errors in my project): 

In file included from C:\Users\sgryb\Documents\Arduino\libraries\STM32SD\src/Sd2Card.h:40,
                 from C:\Users\sgryb\Documents\Arduino\libraries\STM32SD\src/STM32SD.h:22,
                 from C:\Users\sgryb\Documents\Arduino\Libraries\STM32SD\examples\Datalogger\Datalogger.ino:13:
C:\Users\sgryb\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:58:40: error: variable or field 'BSP_SD_GetCardInfo' declared void
   58 | #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef
      |                                        ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\sgryb\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:115:28: note: in expansion of macro 'HAL_SD_CardInfoTypedef'
  115 | void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);
      |                            ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\sgryb\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:58:40: error: 'HAL_SD_CardInfoTypeDef' was not declared in this scope; did you mean 'HAL_SD_CardInfoTypedef'?
   58 | #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef
      |                                        ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\sgryb\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:115:28: note: in expansion of macro 'HAL_SD_CardInfoTypedef'
  115 | void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);
      |                            ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\sgryb\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:115:52: error: 'CardInfo' was not declared in this scope; did you mean 'SD_CardInfo'?
  115 | void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);
      |                                                    ^~~~~~~~
      |                                                    SD_CardInfo
C:\Users\sgryb\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:120:24: error: variable or field 'BSP_SD_MspInit' declared void
  120 | void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                        ^~~~~~~~~~~~~~~~
C:\Users\sgryb\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:120:24: error: 'SD_HandleTypeDef' was not declared in this scope; did you mean 'USBD_HandleTypeDef'?
  120 | void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                        ^~~~~~~~~~~~~~~~
      |                        USBD_HandleTypeDef
C:\Users\sgryb\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:120:42: error: 'hsd' was not declared in this scope
  120 | void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                                          ^~~
C:\Users\sgryb\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:120:47: error: expected primary-expression before 'void'
  120 | void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                                               ^~~~
C:\Users\sgryb\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:121:31: error: variable or field 'BSP_SD_Detect_MspInit' declared void
  121 | void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                               ^~~~~~~~~~~~~~~~
C:\Users\sgryb\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:121:31: error: 'SD_HandleTypeDef' was not declared in this scope; did you mean 'USBD_HandleTypeDef'?
  121 | void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                               ^~~~~~~~~~~~~~~~
      |                               USBD_HandleTypeDef
C:\Users\sgryb\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:121:49: error: 'hsd' was not declared in this scope
  121 | void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                                                 ^~~
C:\Users\sgryb\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:121:54: error: expected primary-expression before 'void'
  121 | void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);
      |                                                      ^~~~
C:\Users\sgryb\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:122:26: error: variable or field 'BSP_SD_MspDeInit' declared void
  122 | void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);
      |                          ^~~~~~~~~~~~~~~~
C:\Users\sgryb\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:122:26: error: 'SD_HandleTypeDef' was not declared in this scope; did you mean 'USBD_HandleTypeDef'?
  122 | void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);
      |                          ^~~~~~~~~~~~~~~~
      |                          USBD_HandleTypeDef
C:\Users\sgryb\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:122:44: error: 'hsd' was not declared in this scope
  122 | void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);
      |                                            ^~~
C:\Users\sgryb\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:122:49: error: expected primary-expression before 'void'
  122 | void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);
      |                                                 ^~~~
C:\Users\sgryb\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:58:40: error: 'HAL_SD_CardInfoTypeDef' does not name a type; did you mean 'HAL_SD_CardInfoTypedef'?
   58 | #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef
      |                                        ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\sgryb\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:64:21: note: in expansion of macro 'HAL_SD_CardInfoTypedef'
   64 | #define SD_CardInfo HAL_SD_CardInfoTypedef
      |                     ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\sgryb\Documents\Arduino\libraries\STM32SD\src/Sd2Card.h:62:5: note: in expansion of macro 'SD_CardInfo'
   62 |     SD_CardInfo _SdCardInfo;
      |     ^~~~~~~~~~~
exit status 1
Error compiling for board Generic STM32F1 series.

fpistm, 2021-01-04:
From README.md:
> This library is for SD card slots connected to the SDIO-/SDMMC-hardware of the processor. For slots connected to SPI-hardware use the standard Arduino SD library.

Not all F1 have SDIO peripheral. Which one you used?

ShaggyDog18, 2021-01-04:
I see... I am using a blue pill with external microSD connected to SPI1

fpistm, 2021-01-04:
So use the Arduino SD library as this F1 does not have SDIO.

closed 2021-01-04 by fpistm

ShaggyDog18, 2021-01-04:
OK, thank you for your prompt reply! 
using SDFat library now, it finally works: https://github.com/greiman/SdFat/releases