stm32duino/STM32SD #36: STM32F407VET6 with SDIO peripheral not compiling STM32SD examples

CognitiveConnexionist, 2021-01-11:
Hello,

after installing STM32SD, when I try to compile the "ReadWrite" example that comes with the STM32SD arduino lib, I get the following errors:

Arduino : 1.8.13 (Windows 8.1), Carte : "Generic STM32F4 series, FK407M1 STM32F407VET, STM32CubeProgrammer (DFU), Enabled (generic 'Serial'), None, Low/Full Speed, Smallest (-Os default), Newlib Nano (default)"

In file included from C:\Users\admin\Documents\Arduino\libraries\STM32SD\src/Sd2Card.h:40,

                 from C:\Users\admin\Documents\Arduino\libraries\STM32SD\src/STM32SD.h:22,

                 from C:\Users\admin\AppData\Local\Temp\arduino_modified_sketch_128243\ReadWrite.ino:12:

C:\Users\admin\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:58:40: error: variable or field 'BSP_SD_GetCardInfo' declared void

   58 | #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef

      |                                        ^~~~~~~~~~~~~~~~~~~~~~

C:\Users\admin\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:115:28: note: in expansion of macro 'HAL_SD_CardInfoTypedef'

  115 | void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);

      |                            ^~~~~~~~~~~~~~~~~~~~~~

C:\Users\admin\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:58:40: error: 'HAL_SD_CardInfoTypeDef' was not declared in this scope; did you mean 'HAL_SD_CardInfoTypedef'?

   58 | #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef

      |                                        ^~~~~~~~~~~~~~~~~~~~~~

C:\Users\admin\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:115:28: note: in expansion of macro 'HAL_SD_CardInfoTypedef'

  115 | void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);

      |                            ^~~~~~~~~~~~~~~~~~~~~~

C:\Users\admin\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:115:52: error: 'CardInfo' was not declared in this scope; did you mean 'SD_CardInfo'?

  115 | void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);

      |                                                    ^~~~~~~~

      |                                                    SD_CardInfo

C:\Users\admin\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:120:24: error: variable or field 'BSP_SD_MspInit' declared void

  120 | void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);

      |                        ^~~~~~~~~~~~~~~~

C:\Users\admin\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:120:24: error: 'SD_HandleTypeDef' was not declared in this scope; did you mean 'SPI_HandleTypeDef'?

  120 | void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);

      |                        ^~~~~~~~~~~~~~~~

      |                        SPI_HandleTypeDef

C:\Users\admin\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:120:42: error: 'hsd' was not declared in this scope

  120 | void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);

      |                                          ^~~

C:\Users\admin\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:120:47: error: expected primary-expression before 'void'

  120 | void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);

      |                                               ^~~~

C:\Users\admin\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:121:31: error: variable or field 'BSP_SD_Detect_MspInit' declared void

  121 | void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);

      |                               ^~~~~~~~~~~~~~~~

C:\Users\admin\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:121:31: error: 'SD_HandleTypeDef' was not declared in this scope; did you mean 'SPI_HandleTypeDef'?

  121 | void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);

      |                               ^~~~~~~~~~~~~~~~

      |                               SPI_HandleTypeDef

C:\Users\admin\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:121:49: error: 'hsd' was not declared in this scope

  121 | void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);

      |                                                 ^~~

C:\Users\admin\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:121:54: error: expected primary-expression before 'void'

  121 | void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);

      |                                                      ^~~~

C:\Users\admin\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:122:26: error: variable or field 'BSP_SD_MspDeInit' declared void

  122 | void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);

      |                          ^~~~~~~~~~~~~~~~

C:\Users\admin\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:122:26: error: 'SD_HandleTypeDef' was not declared in this scope; did you mean 'SPI_HandleTypeDef'?

  122 | void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);

      |                          ^~~~~~~~~~~~~~~~

      |                          SPI_HandleTypeDef

C:\Users\admin\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:122:44: error: 'hsd' was not declared in this scope

  122 | void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);

      |                                            ^~~

C:\Users\admin\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:122:49: error: expected primary-expression before 'void'

  122 | void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);

      |                                                 ^~~~

C:\Users\admin\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:58:40: error: 'HAL_SD_CardInfoTypeDef' does not name a type; did you mean 'HAL_SD_CardInfoTypedef'?

   58 | #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef

      |                                        ^~~~~~~~~~~~~~~~~~~~~~

C:\Users\admin\Documents\Arduino\libraries\STM32SD\src/bsp_sd.h:64:21: note: in expansion of macro 'HAL_SD_CardInfoTypedef'

   64 | #define SD_CardInfo HAL_SD_CardInfoTypedef

      |                     ^~~~~~~~~~~~~~~~~~~~~~

C:\Users\admin\Documents\Arduino\libraries\STM32SD\src/Sd2Card.h:62:5: note: in expansion of macro 'SD_CardInfo'

   62 |     SD_CardInfo _SdCardInfo;

      |     ^~~~~~~~~~~

exit status 1

Erreur de compilation pour la carte Generic STM32F4 series

fpistm, 2021-01-11:
Hi @CognitiveConnexionist 
this is normal.
The variant does not enable the HAL SD module that's why you have all those errors.
Moreover the PinMap SD does not exists in the PeripheralPins.c.

You can enable the `HAL_SD_MODULE_ENABLED` by adding a file at the sketch level name `hal_conf_extra.h` with:
`#define HAL_SD_MODULE_ENABLED`

For the missing array, add on top of your sketch this array:
```c
#ifdef HAL_SD_MODULE_ENABLED
const PinMap PinMap_SD[] = {
  // {PB_8,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D4
  // {PB_9,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D5
  // {PC_6,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D6
  // {PC_7,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D7
  {PC_8,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D0
  {PC_9,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D1
  {PC_10, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D2
  {PC_11, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D3
  {PC_12, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, GPIO_AF12_SDIO)}, // SDIO_CK
  {PD_2,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, GPIO_AF12_SDIO)}, // SDIO_CMD
  {NC,    NP,    0}
};
#endif
```

You can comment or remove the pin you don't need.

closed 2021-01-11 by fpistm

Semsem-220109, 2023-10-11:
Hi @CognitiveConnexionist 
I'd like to know how did you define the pins for the SDIO connection and did it work?