stm32duino/STM32SD #37: Canot compile, linker errors

Bambofy, 2021-01-27:
Hi i'm trying to use the STM32SD library but i can't get it to compile.

Here's the linker messages:
```
"Error linking for board Nucleo-64
Build failed for project 'SDMMC_Test'
ld.exe: STM32SD-1.2.2\bsp_sd.c.o: in function BSP_SD_MspInit
bsp_sd.c:435: undefined reference to PinMap_SD
 
ld.exe: STM32SD-1.2.2\bsp_sd.c.o: in function BSP_SD_Init
bsp_sd.c:163: undefined reference to HAL_SD_Init
ld.exe: C:\Users\Richa\Documents\Arduino\libraries\STM32SD-1.2.2\src\bsp_sd.c:174: undefined reference to HAL_SD_ConfigWideBusOperation
 
ld.exe: STM32SD-1.2.2\bsp_sd.c.o: in function BSP_SD_GetCardInfo
bsp_sd.c:557: undefined reference to HAL_SD_GetCardInfo
 
collect2.exe*: error: ld returned 1 exit status"
```

I've used these flags in the hal_conf_extra.h which at the root of my project.
```
#define HAL_MODULE_ENABLED
#define HAL_SAI_MODULE_ENABLED
#define HAL_TIM_MODULE_ENABLED
#define USE_FULL_ASSERT
#define HAL_ADC_MODULE_ENABLED
#define HAL_MMC_MODULE_ENABLED
#define HAL_SD_MODULE_ENABLED
#define HAL_GPIO_MODULE_ENABLED
#define HAL_EXTI_MODULE_ENABLED
#define HAL_I2C_MODULE_ENABLED
#define HAL_DMA_MODULE_ENABLED
#define HAL_RCC_MODULE_ENABLED
#define HAL_FLASH_MODULE_ENABLED
#define HAL_PWR_MODULE_ENABLED
#define HAL_CORTEX_MODULE_ENABLED
```

fpistm, 2021-01-27:
Hi @Bambofy 
which target you used ?

Bambofy, 2021-01-27:
> 
> 
> Hi @Bambofy
> which target you used ?

Nucleo L452RE

fpistm, 2021-01-27:
Not the L452RE-P?

Bambofy, 2021-01-27:
No

fpistm, 2021-01-27:
Could you be more precise on your environment? 
Arduino IDE version? STM32 core version? STM32SD library version? ...
Thanks in advance

fpistm, 2021-01-27:
Using the latest Core release 19.0, the latest STM32SD 1.2.2, Arduino IDE 1.8.13 and enabling the HAL SD module thanks the hal_conf_extra.h. I'm able to build properly the CardInfo example:

> Sketch uses 31252 bytes (5%) of program storage space. Maximum is 524288 bytes.
> Global variables use 2304 bytes (1%) of dynamic memory, leaving 161536 bytes for local variables. Maximum is 163840 bytes.
>

Bambofy, 2021-01-27:
Sorry for the delay,

Core version is 1.9.0 from the  library manager, i am using visual micro to upload the code, when i try arduino IDE instead, it works so it must be a bug with visual micro. Thanks for your help :)!

closed 2021-01-27 by Bambofy

fpistm, 2021-01-28:
I'm using also VSCode to debug but I do not have this issue. So probably a wrong configuration with Arduino plugin.