stm32duino/STM32SD #39: Appending to an existing file?

Bambofy, 2021-02-01:
Hi, i'm trying to append data to an already existing file, i see that FatFS has the flags (defined in ff.h):

```
/* File access mode and open method flags (3rd argument of f_open) */
#define	FA_READ				0x01
#define	FA_WRITE			0x02
#define	FA_OPEN_EXISTING	0x00
#define	FA_CREATE_NEW		0x04
#define	FA_CREATE_ALWAYS	0x08
#define	FA_OPEN_ALWAYS		0x10
#define	FA_OPEN_APPEND		0x30
```

And i think i need to use the FA_OPEN_APPEND flag, is it compatible with the SD.open() function?
`
File f = SD.open(inFileName, FA_OPEN_APPEND);`

fpistm, 2021-02-01:
Hi @Bambofy 
yes it should be OK, at least the open will be done in the correct mode as it simply call `f_open` with the mode gave for `SD.open()`.
The easiest way is to try. 😉

fpistm, 2021-02-01:
Please use the forum for support or question.

closed 2021-02-01 by fpistm