stm32duino/STM32SD #41: Broken with 1.9.0 +

BennehBoy, 2021-03-27:
Hi,

Just tried building one of my projects that requies this library against the github release of the core and get the following:

`In file included from C:\Users\BennehBoy\Documents\Arduino\libraries\STM32duino_STM32SD\src/Sd2Card.h:40,

                 from C:\Users\BennehBoy\Documents\Arduino\libraries\STM32duino_STM32SD\src/STM32SD.h:22,

                 from C:\Users\BennehBoy\Documents\Arduino\LRDuinoTD5\LRduinoV0.018 - Copy\LRDuinoTD5\LRDuinoTD5.ino:41:

C:\Users\BennehBoy\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:58:40: error: variable or field 'BSP_SD_GetCardInfo' declared void

   58 | #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef

      |                                        ^~~~~~~~~~~~~~~~~~~~~~

C:\Users\BennehBoy\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:115:28: note: in expansion of macro 'HAL_SD_CardInfoTypedef'

  115 | void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);

      |                            ^~~~~~~~~~~~~~~~~~~~~~

C:\Users\BennehBoy\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:58:40: error: 'HAL_SD_CardInfoTypeDef' was not declared in this scope; did you mean 'HAL_SD_CardInfoTypedef'?

   58 | #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef

      |                                        ^~~~~~~~~~~~~~~~~~~~~~

C:\Users\BennehBoy\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:115:28: note: in expansion of macro 'HAL_SD_CardInfoTypedef'

  115 | void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);

      |                            ^~~~~~~~~~~~~~~~~~~~~~

C:\Users\BennehBoy\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:115:52: error: 'CardInfo' was not declared in this scope; did you mean 'SD_CardInfo'?

  115 | void    BSP_SD_GetCardInfo(HAL_SD_CardInfoTypedef *CardInfo);

      |                                                    ^~~~~~~~

      |                                                    SD_CardInfo

C:\Users\BennehBoy\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:120:24: error: variable or field 'BSP_SD_MspInit' declared void

  120 | void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);

      |                        ^~~~~~~~~~~~~~~~

C:\Users\BennehBoy\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:120:24: error: 'SD_HandleTypeDef' was not declared in this scope; did you mean 'USBD_HandleTypeDef'?

  120 | void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);

      |                        ^~~~~~~~~~~~~~~~

      |                        USBD_HandleTypeDef

C:\Users\BennehBoy\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:120:42: error: 'hsd' was not declared in this scope

  120 | void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);

      |                                          ^~~

C:\Users\BennehBoy\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:120:47: error: expected primary-expression before 'void'

  120 | void    BSP_SD_MspInit(SD_HandleTypeDef *hsd, void *Params);

      |                                               ^~~~

C:\Users\BennehBoy\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:121:31: error: variable or field 'BSP_SD_Detect_MspInit' declared void

  121 | void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);

      |                               ^~~~~~~~~~~~~~~~

C:\Users\BennehBoy\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:121:31: error: 'SD_HandleTypeDef' was not declared in this scope; did you mean 'USBD_HandleTypeDef'?

  121 | void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);

      |                               ^~~~~~~~~~~~~~~~

      |                               USBD_HandleTypeDef

C:\Users\BennehBoy\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:121:49: error: 'hsd' was not declared in this scope

  121 | void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);

      |                                                 ^~~

C:\Users\BennehBoy\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:121:54: error: expected primary-expression before 'void'

  121 | void    BSP_SD_Detect_MspInit(SD_HandleTypeDef *hsd, void *Params);

      |                                                      ^~~~

C:\Users\BennehBoy\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:122:26: error: variable or field 'BSP_SD_MspDeInit' declared void

  122 | void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);

      |                          ^~~~~~~~~~~~~~~~

C:\Users\BennehBoy\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:122:26: error: 'SD_HandleTypeDef' was not declared in this scope; did you mean 'USBD_HandleTypeDef'?

  122 | void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);

      |                          ^~~~~~~~~~~~~~~~

      |                          USBD_HandleTypeDef

C:\Users\BennehBoy\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:122:44: error: 'hsd' was not declared in this scope

  122 | void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);

      |                                            ^~~

C:\Users\BennehBoy\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:122:49: error: expected primary-expression before 'void'

  122 | void    BSP_SD_MspDeInit(SD_HandleTypeDef *hsd, void *Params);

      |                                                 ^~~~

C:\Users\BennehBoy\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:58:40: error: 'HAL_SD_CardInfoTypeDef' does not name a type; did you mean 'HAL_SD_CardInfoTypedef'?

   58 | #define HAL_SD_CardInfoTypedef         HAL_SD_CardInfoTypeDef

      |                                        ^~~~~~~~~~~~~~~~~~~~~~

C:\Users\BennehBoy\Documents\Arduino\libraries\STM32duino_STM32SD\src/bsp_sd.h:64:21: note: in expansion of macro 'HAL_SD_CardInfoTypedef'

   64 | #define SD_CardInfo HAL_SD_CardInfoTypedef

      |                     ^~~~~~~~~~~~~~~~~~~~~~

C:\Users\BennehBoy\Documents\Arduino\libraries\STM32duino_STM32SD\src/Sd2Card.h:62:5: note: in expansion of macro 'SD_CardInfo'

   62 |     SD_CardInfo _SdCardInfo;

      |     ^~~~~~~~~~~
`

I've tried this against the github release & 1.9.0 and get the same.

fpistm, 2021-03-27:
This simply means you don't have defined the HAL SD MODULE support.
Not all the board have a SD card slot so not all have the definition in the variant.
Use the hat_conf_extra.h and add:
`#define HAL_SD_MODULE_ENABLED`

BennehBoy, 2021-03-27:
Ah, I see what's happened, looks like the fk407 variant that I created which included SD support had it removed for some reason.  Good spot.

fpistm, 2021-03-27:
So it can be closed ?

BennehBoy, 2021-03-27:
I updated hal_extra_conf.h

But I guess there's another issue with the FK407 variant

c:/users/bennehboy/appdata/local/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\BENNEH~1\AppData\Local\Temp\arduino_build_484934\libraries\STM32duino_STM32SD\bsp_sd.c.o: in function `BSP_SD_MspInit':
bsp_sd.c:(.text.BSP_SD_MspInit+0x34): undefined reference to `PinMap_SD'

I'll go dig

BennehBoy, 2021-03-27:
> So it can be closed ?

Yes I think it can be closed here - I'll raise an issue against the variant (or PR).

BennehBoy, 2021-03-27:
OK easy fix - PeripheralPins.c for FK407 misses this:

#ifdef HAL_SD_MODULE_ENABLED
WEAK const PinMap PinMap_SD[] = {
  // {PB_8,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D4
  // {PB_9,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D5
  // {PC_6,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D6
  // {PC_7,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D7
  {PC_8,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D0
  {PC_9,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D1
  {PC_10, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D2
  {PC_11, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D3
  {PC_12, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, GPIO_AF12_SDIO)}, // SDIO_CK
  {PD_2,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, GPIO_AF12_SDIO)}, // SDIO_CMD
  {NC,    NP,    0}
};
#endif

BennehBoy, 2021-03-27:
https://github.com/stm32duino/Arduino_Core_STM32/pull/1337

fpistm, 2021-03-27:
This will be superseded by https://github.com/stm32duino/Arduino_Core_STM32/pull/1091

closed 2021-03-27 by fpistm