stm32duino/STM32SD #43: Please publish a new version with the patch to the datalogger.ino example

jeback1, 2021-04-01:
I used the datalogger example as the basis for logging sensor data in my project. I had the problems Arnas MasRodjie described in his patch: "Fix Datalogger.ino example fragments the memory because of loop opening & closing the file. This will make the memory not able to allocate anymore. Keep the file opened and use flush to write the data instead opening & closing the file."
His patch using flush instead of repeatedly opening and closing appears to have solved the problems I was having with eventually having an error when trying to open the file and ultimately the program stops printing to Serial. 

The version provided in the Arduino IDE (v 1.2.2) doesn't include his patch. I suggest updating the release to include this fix.

Thank you.
Jan Erik

fpistm, 2021-04-01:
Hi @jeback1
In fact the rework of the datalogger fix(hide) the sketch issue but not the real issue raised by @AnHardt in #24. He propose a fix that I have to review in #31.

jeback1, 2021-04-02:
Very good.  Thanks for the elaboration.

fpistm, 2021-04-22:
Done --> https://github.com/stm32duino/STM32SD/releases/tag/1.2.3

closed 2021-04-22 by fpistm