stm32duino/STM32SD #46: Microcontroller freezes

Gurkengewuerz, 2021-06-15:
Hey,

i tried this library on a STM32F411 ("Blackpill" by Adafruit _4877_). I've connected a SDHC card to it using the 1-Bit mode.
But all it does is freezing in the initialization process.

```
uSD CMD => PA06
uSD CLCK => PB15
uSD D0 => PB04
uSD D1 => PA08
```

After some testing i could figure out, that during the sd initialisation this part cause the issue
https://github.com/stm32duino/STM32SD/blob/32a42605b16c8b05749ca875c61faf0fed77cbf7/src/bsp_sd.c#L172-L179

Afterwards my SDHC card is recognized but the fatfs could not be initialized. So i guess there is some configuration issue?

Gurkengewuerz, 2021-06-15:
Update: Same happens in 4-Bit Mode!

fpistm, 2021-06-15:
Hi @Gurkengewuerz 
Could you be more precise about your environment?
STM32 core version?
Arduino IDE version?
STM32SD library version?
FatFas library version?
Target selected in the menu?

Gurkengewuerz, 2021-06-15:
Hey @fpistm thanks for your fast feedback!
Sure.

I'm using PlatformIO with the newest available STM32 core.
```ini
[common]
monitor_speed = 9600
build_flags = -D MON_SPEED=${common.monitor_speed}
lib_deps =
            https://github.com/stm32duino/FatFs#cc9d17d
            https://github.com/stm32duino/STM32SD#8dc2821

[env:genericSTM32F411CC]
platform = ststm32
board = genericSTM32F411CC
board_build.f_cpu = 100000000L
upload_protocol = stlink
framework = arduino
lib_deps = ${common.lib_deps}
monitor_speed = ${common.monitor_speed}
build_flags = ${common.build_flags}
```

```
Package framework-arduinoststm32
--------------------------------
Type: framework
Requirements: ~4.20000.0
Installed: Yes
Version: 4.20000.210603
Original version: 2.0.0
Description: Arduino Wiring-based Framework for ST STM32 microcontrollers
```

The SD Lib is the newest commit https://github.com/stm32duino/STM32SD#8dc282. Same as for the FatFS https://github.com/stm32duino/FatFs#cc9d17d

Gurkengewuerz, 2021-06-15:
To be even more precice on the "freeze".
I thought it freezes, but after around 3 Minutes the begin function timed out (if a SD Card is present).
If i remove the sd card, the begin functions return false immediately.

fpistm, 2021-06-15:
Thanks for the feedback.
We don't support PIO only Arduino IDE.
For example you define: `board_build.f_cpu = 100000000L`
OK the boards can run at this freq anyway the generic F411CC (note that Adafruit board have a CE) by default use HSI as clock source and run at 96 MHz in order to have USB clock at 48MHz so f_cpu is not correct. This could cause some timing issue.

Note that by default `SD_BUS_WIDE` is defined at `SD_BUS_WIDE_4B`. I guess you know how manage it as you have tested both.

I've tested the SDIO with a F411 and it works as expected with Arduino IDE. 
So I close this one as I will not able to help with PIO.

closed 2021-06-15 by fpistm

Gurkengewuerz, 2021-06-15:
Hmm okay. I guess i try changing f_cpu. But to be sure: since the uC provides multiple SDIO_DX pins, does the library care which one I use?

fpistm, 2021-06-15:
Currently the library configure all pins avialble in the `PinMap_SD` array here: https://github.com/stm32duino/Arduino_Core_STM32/blob/03052da8712de2e2f48b0460da079ff3c50e3fb0/variants/STM32F4xx/F411C(C-E)(U-Y)/PeripheralPins.c#L237-L250

It is planned to split this table per SD signal and be able to define only one as it has been done for QSPI/OSPI.
Anyway you can redefine this array to only provide the one you need:
https://github.com/stm32duino/wiki/wiki/Custom-definitions#custom-pinmap-array

fpistm, 2021-06-15:
But as said we do not support PIO so don't know how this is handled...

pavel-d, 2024-09-15:
Hey @Gurkengewuerz, I have exactly the same problem. Have you figured it out?

pavel-d, 2024-09-15:
@fpistm, I tried to reproduce it using the [CardInfo](https://github.com/stm32duino/STM32SD/blob/75cded50f0da6c7e47eb01c52df62588a7618908/examples/CardInfo/CardInfo.ino) example Arduino, it freezes anyway. Any ideas how to debug it further?

```
14:05:42.317 -> 
14:05:42.317 -> Initializing SD card...A card is present.
14:05:42.317 -> 
14:05:42.317 -> Card type: SDHC
```
_(it just hangs here)_

I'm using 1-bit wide data bus on standard pins on STM32F722RET6. SD card is formatted with FAT32 and is readable on other devices. The only change I made is defined `#define SD_BUS_WIDE SD_BUS_WIDE_1B`

When I remove the SD card it doesn't hang (but obviously fails to initialise the fatFs).

fpistm, 2024-09-16:
Hi @pavel-d 
I've tested the SD_BUS_WIDE_1B with a DISCO F746NG and no issue.
Your error seems that the file system is not correctly detected or mounted.