stm32duino/STM32SD #47: How are multiple pin definitions handled?

sakza, 2021-07-26:
Not sure if this belongs here or in the stm32 core repository - for the f411ceux pin definition, there are two pins defined for SDIO_D0, PB4 and PB7. This seems consistent with the f411 documentation, so obviously not an error/typo. This SD library didn't work, however, until I overwrote the SD pin definition with only a single pin defined as D0.

Is this the correct behaviour? If so, some sort of compiler warning/error could be useful in similar scenarios with other chips.

sakza, 2021-07-26:
Here is the referenced code from the [STM32F4xx/F411C(C-E)(U-Y)/PeripheralPins.c](https://github.com/stm32duino/Arduino_Core_STM32/blob/master/variants/STM32F4xx/F411C(C-E)(U-Y)/PeripheralPins.c):

`#ifdef HAL_SD_MODULE_ENABLED
WEAK const PinMap PinMap_SD[] = {
  {PA_6,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, GPIO_AF12_SDIO)}, // SDIO_CMD
  {PA_8,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D1
  {PA_9,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D2
  {PB_4,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D0
  {PB_5,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D3
  {PB_7,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D0
  {PB_8,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D4
  {PB_9,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D5
  {PB_10, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D7
  {PB_14, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D6
  {PB_15, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, GPIO_AF12_SDIO)}, // SDIO_CK
  {NC,    NP,   0}
};
#endif`

fpistm, 2021-08-13:
Hi @sakza 
Sorry for the delay, I was on vacation.
You are right. All pins defined in the PinMap_SD array are configured that's why you see this issue.
As we are not able to know which one have to be used the correct way is to redefine it as you've made.

Anyway, I've planned to split the array per signal and then provide a way to define each pins to used as I've made for QSPI/OSPI