stm32duino/STM32SD #59: Generic STM32F407ZG and STM32H743IIT6 problem - SDIO not working...

Mobin-Mortazavi, 2023-06-24:
Hi

I tested the Example "CardInfo" with both of them and after a while it get out of card.init() loop and says : "initialization failed. Is a card inserted?"

I tested them with defaults and I even Changed pins with given commands in the .md file of yours

about my hardware , It's a development board and I tested it with stm32cube and it worked!

any Idea ?!

fpistm, 2023-06-25:
Which core version you used? and STM32SD version?
The new API will be available within the next core release 2.6.0

Mobin-Mortazavi, 2023-06-25:
Thank you so much... 
You are  developing super useful libraries and it's great! 💯 

STM32 core version : 2.5.0
STM32SD version : 1.3

fpistm, 2023-06-26:
Which pins you used? You also have to ensure clock config fot SD is 48Mhz. It's work for those mcu.

Mobin-Mortazavi, 2023-06-26:
> Which pins you used? You also have to ensure clock config for SD is 48Mhz. It's work for those MCUs

The Hardware pins for the SDIO on both of them are the same => SDIO1 ( PC8 , PC9 , PC10 , PC11 , PD2 , PC12 )

How should I Change or Check the Clock? Should I override HAL Clock config method ?
when I change clocks using  HAL commands in the Setup() my Serial and other peripherals stop working.

fpistm, 2023-06-27:
> The Hardware pins for the SDIO on both of them are the same => SDIO1 ( PC8 , PC9 , PC10 , PC11 , PD2 , PC12 )

No exactly, for F4 it is `SDIO `for H7 it is `SDMMC1`.

To change the system core clock config, simply redefine it at sketch level with `extern "C"` as explained [here](https://github.com/stm32duino/Arduino_Core_STM32/wiki/Custom-definitions#systemclock_config):
```
extern "C" void SystemClock_Config(void) {
}
```

With core 2.5.0 you can redefine the `PinMap_SD` array to remove useless pins as explained [here](https://github.com/stm32duino/Arduino_Core_STM32/wiki/Custom-definitions#example-for-the-adc-pinmap-of-the-nucleo_f103rb):
```
const PinMap PinMap_SD[] = {
  // {PB_8,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D4
  // {PB_9,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D5
  // {PC_6,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D6
  // {PC_7,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D7
  {PC_8,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D0
  {PC_9,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D1
  {PC_10, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D2
  {PC_11, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D3
  {PC_12, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, GPIO_AF12_SDIO)}, // SDIO_CK
  {PD_2,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, GPIO_AF12_SDIO)}, // SDIO_CMD
  {NC,    NP,   0}
};
```

And mainly for H7 which have more pins.

Honestly, I don't know how to help you more. I've tested and it works with those mcu. It is linked to your hardware and/or software configuration.

Mobin-Mortazavi, 2023-06-27:
Thanks a lot but yet I can't make it work.

I attach the schematics , But I removed that hardware pull-ups because I noticed that you did the pulling in the software
 
[Schematic-F400Z-RevB_2.pdf](https://github.com/stm32duino/STM32SD/files/11883053/Schematic-F400Z-RevB_2.pdf)

fpistm, 2023-06-27:
I guess you configure the HSE as PLL source but seems you have a 25 MHz while by default it is 8MHz so you have to redefine it to 25MHz:
https://github.com/stm32duino/Arduino_Core_STM32/blob/c6bc5b23761c30ff5ef08baa22af1519b0a76d5a/system/STM32F4xx/stm32f4xx_hal_conf_default.h#L100

To do that, add a file name `build_opt.h` and add: `-DHSE_VALUE=25000000U`
WiKi for more information: https://github.com/stm32duino/Arduino_Core_STM32/wiki/Customize-build-options-using-build_opt.h

Then add this in your sketch:

```
extern "C" void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {};

  /** Configure the main internal regulator output voltage
  */
  __HAL_RCC_PWR_CLK_ENABLE();
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);

  /** Initializes the RCC Oscillators according to the specified parameters
  * in the RCC_OscInitTypeDef structure.
  */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSE;
  RCC_OscInitStruct.HSEState = RCC_HSE_ON;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
  RCC_OscInitStruct.PLL.PLLM = 25;
  RCC_OscInitStruct.PLL.PLLN = 336;
  RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
  RCC_OscInitStruct.PLL.PLLQ = 7;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)  {
    Error_Handler();
  }

  /** Initializes the CPU, AHB and APB buses clocks
  */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK | RCC_CLOCKTYPE_SYSCLK
                                                 | RCC_CLOCKTYPE_PCLK1 | RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV4;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV2;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_5) != HAL_OK) {
    Error_Handler();
  }
}
```

fpistm, 2023-07-10:
Considered resolved. No issue on those targets. User configuration issue.

closed 2023-07-10 by fpistm