stm32duino/STM32SD #60: STEVAL-MKSBOX1V1 SDIO broken under STMduino STM32SD 1.3.0; works under 1.2.3

gemelko, 2023-07-05:
A new project that I created using the STM32duino STM32SD library would not access any SD card on the sensortile box (STEVAL-MKSBOX1V1). I also had a previous project which used to work however was now failing. 

So then tried several of the included examples (listfiles, CardInfo, etc) and I found that none of them worked either. It does one of two things: hangs for a long time then fails on the SD.begin() call, or in the case of CardInfo, it sometimes passes the card.init() call but then fails the fatFs.init() call and reports that the card is not formatted. I tried this with several different SD cards, formatted to either FAT16 or FAT32: all had the same problem.

I realized that I had upgraded to library version 1.3.0 recently, and so I removed 1.3.0 and installed 1.2.3. Now all of the examples and my code work properly for all cards. I did not dig into the library or causes any further.

Glenn

fpistm, 2023-07-05:
Hi @gemelko 
Thanks for the report.
I've made a fix in #61.

gemelko, 2023-07-05:
I can confirm that this works properly. Thanks!

fpistm, 2023-07-05:
> I can confirm that this works properly. Thanks!

Anyway, I've found an other issue that I will fix for other series due to my mistakes 😩

PR #61 (2023-07-07): fix: wrong preprocessor definitions
Fixes #60
commit 338f6007 (2023-07-07): fix: restore SD_CLK_ENABLE

Wrong endif preprocessor placement when introducing new
SD pins management.

Fixes #60
closed 2023-07-07 by fpistm