stm32duino/STM32SD #80: STM32SD Speed Test Example

ywf1, 2024-11-23:
If possible could an example be added or provided with a read / write speed figure - currently using a stm32f405 with sdio interface, I want to identify maximum sampling rate of sensors, so wanted a good idea of the read/write (mostly write) speed using the lib.

Thank you for the help!

fpistm, 2024-11-25:
Hi @ywf1 
That's a good idea but have no time soon. Any contribution are welcome.

ywf1, 2024-11-30:
Dear Fredric,

Tried it out myself!

Hardware: STM32F405, SDIO Interface
SD Card: SanDisk Ultra 32GB MicroSD HC I

Script:
`#include <STM32SD.h>

#define TEST_FILE_NAME "/speed_test.txt"
#define TEST_BLOCK_SIZE 4096  // Size of each data block in bytes
#define TEST_BLOCK_COUNT 400 // Number of blocks to write and read

File testFile;

void setup() {
  delay(2000);

  Serial.begin(115200);
  while (!Serial) {
    delay(10); // Wait for Serial to initialize
  }

  Serial.println("Initializing SD card...");
  if (!SD.begin()) {
    Serial.println("SD initialization failed!");
    while (1);
  }
  Serial.println("SD initialization successful.");

  // Perform write speed test
  writeSpeedTest();

  // Perform read speed test
  readSpeedTest();
}

void writeSpeedTest() {
  Serial.println("Starting write speed test...");

  // Prepare test data
  uint8_t buffer[TEST_BLOCK_SIZE];
  for (int i = 0; i < TEST_BLOCK_SIZE; i++) {
    buffer[i] = i % 256;
  }

  // Open file for writing
  testFile = SD.open(TEST_FILE_NAME, FILE_WRITE);
  if (!testFile) {
    Serial.println("Failed to open file for writing.");
    return;
  }

  unsigned long startTime = micros();
  for (int i = 0; i < TEST_BLOCK_COUNT; i++) {
    testFile.write(buffer, TEST_BLOCK_SIZE);
  }
  testFile.close();
  unsigned long elapsedTime = micros() - startTime;

  float speed = (TEST_BLOCK_SIZE * TEST_BLOCK_COUNT) / (elapsedTime / 1000000.0) / 1024.0;
  Serial.print("Write speed: ");
  Serial.print(speed, 2); // Display speed with 2 decimal places
  Serial.println(" KB/s");
}

void readSpeedTest() {
  Serial.println("Starting read speed test...");

  // Open file for reading
  testFile = SD.open(TEST_FILE_NAME, FILE_READ);
  if (!testFile) {
    Serial.println("Failed to open file for reading.");
    return;
  }

  uint8_t buffer[TEST_BLOCK_SIZE];

  unsigned long startTime = micros();
  for (int i = 0; i < TEST_BLOCK_COUNT; i++) {
    testFile.read(buffer, TEST_BLOCK_SIZE);
  }
  testFile.close();
  unsigned long elapsedTime = micros() - startTime;

  float speed = (TEST_BLOCK_SIZE * TEST_BLOCK_COUNT) / (elapsedTime / 1000000.0) / 1024.0;
  Serial.print("Read speed: ");
  Serial.print(speed, 2); // Display speed with 2 decimal places
  Serial.println(" KB/s");
}

void loop() {
  // Do nothing in loop
}
`
Results: 
[image]

Do these seem reasonable? 

Best,
Yahya