stm32duino/STM32SD #83: STM32L5 Crash while reading SD card or calling .end()

haydenth, 2025-03-27:
I'm experiencing an issue with the SD card reader and the STM32L5. This is using the STM32 DK board referenced here:
https://github.com/stm32duino/Arduino_Core_STM32/issues/2694

The issue is that reading from the SD card causes the microcontroller to either freeze, or crash and restart if the watchdog is enabled (i have watchdog enabled).

**To Reproduce a Crash**
```
#include <Arduino.h>
#include <STM32SD.h>
#include "stm32l5xx_hal.h"
#include "stm32l5xx_hal_flash.h"
#include "stm32l5xx_hal_rcc.h"

#define FIRMWARE_FILENAME "stm32l5_series_test.ino.bin"

File root;
uint32_t loopCount = 0;

void setup() {

  Serial.begin(115200);
  while(!Serial);
  pinMode(LED_GREEN, OUTPUT);
  pinMode(LED_RED, OUTPUT);

  delay(1000);
  Serial.print("Initializing SD card...");

  if (SD.begin(SD_DETECT_PIN)) {
    Serial.println("SD card initialized");
  } else {
    Serial.println("SD card failed to initialize");
  }

}

void loop() {

  digitalWrite(LED_GREEN, HIGH);
  digitalWrite(LED_RED, LOW);
  Serial.println("Hello World");
  delay(1000);
  digitalWrite(LED_GREEN, LOW);
  digitalWrite(LED_RED, HIGH);

  File test = SD.open(FIRMWARE_FILENAME, FILE_READ);
  if (test) {
    Serial.println("Firmware file exists");
    test.close();
  } else {
    Serial.println("No firmware file found");
  }

  loopCount++;
  Serial.print("Loop count: ");
  Serial.println(loopCount);

  Serial.flush();
  Serial.println("Looping in 5 seconds...");
  delay(5000);
}
```

And here's what the serial log looks like as this runs.
```
Initializing SD card...SD card initialized

Hello World

No firmware file found

Loop count: 1

Hello World

No firmware file found

Loop count: 2

Hello World

No firmware file found

Loop count: 3

Hello World

No firmware file found

Loop count: 4

Hello World

Device /dev/ttyACM1 disconnected. Waiting for reconnect...
Device /dev/ttyACM1 not found. Retrying in 1 second...
Device /dev/ttyACM1 detected. Opening connection...
Initializing SD card...SD card initialized

Hello World

No firmware file found

Loop count: 1

Hello World

No firmware file found

Loop count: 2

Hello World

No firmware file found

Loop count: 3
```

So it appears to be crashing every few SD card reads - it's really unpredictable. Sometimes it goes for a while and sometimes it will only do 2-3 loops before there's a crash.

Here's how I build an upload

``
#!/bin/bash

# delete the build folder
rm -rf ../test_build

# compile the code
arduino-cli compile --fqbn STMicroelectronics:stm32:Disco:pnum=STM32L562E_DK,usb=CDCgen,opt=ogstd,dbg=enable_all,rtlib=nanofp \
  /home/tom/projects/corrosion-sensor-firmware/stm32l5_series_test/ \
  --output-dir /home/tom/projects/corrosion-sensor-firmware/test_build \
  --verbose \
  --export-binaries

# if previous command fails, exit
if [ $? -ne 0 ]; then
  exit 1
fi

# erase the memory under Reset
STM32_Programmer_CLI -c port=SWD mode=UR -e all -ob RDP=0xAA

# now upload with hard reset
STM32_Programmer_CLI -c port=SWD mode=UR \
  -d ../test_build/stm32l5_series_test.ino.bin 0x08000000 \
  -v -hardRst
```

And to reproduce a crash even simpler:
```

void setup() {

  Serial.begin(115200);
  while(!Serial);
  pinMode(LED_GREEN, OUTPUT);
  pinMode(LED_RED, OUTPUT);

  delay(1000);
  Serial.print("Initializing SD card...");

  if (SD.begin(SD_DETECT_PIN)) {
    Serial.println("SD card initialized");
  } else {
    Serial.println("SD card failed to initialize");
  }

  SD.end();
}

```

fpistm, 2025-03-27:
Using your sketch, I have no crash.

haydenth, 2025-03-27:
Welp, the issue must be on my end then, I'll push some buttons until it works. Thanks for your help

PR #84 (2025-03-27): fix: wrong GPIO port conversion
Fixes #83
commit 25ee4ac4 (2025-03-27): fix: wrong GPIO port conversion

Fixes #83
closed 2025-03-27 by fpistm