stm32duino/STM32SD #85: Simultaneous use of an SD card and a SPI (QSPI) flash

acronis2000, 2025-07-19:
If you use an SD card or SPI(QSPI) in the same project, the sketch compiles. 
If you need to use both an SD card and SPI(QSPI) in the project, you will get a compilation error

`
#include <SPI.h>
#include <STM32SD.h>  
#include <Adafruit_SPIFlash.h> 
Adafruit_FlashTransport_SPI flashTransport(SS, SPI);  // Set CS and SPI interface
Adafruit_SPIFlash flash(&flashTransport);

In file included from c:\Users\Admin\Documents\Arduino\libraries\STM32duino_STM32SD\src/STM32SD.h:22,
                 from C:\Users\Admin\Documents\Arduino\***.ino:32:
c:\Users\Admin\Documents\Arduino\libraries\STM32duino_STM32SD\src/Sd2Card.h:45:31: error: expected unqualified-id before numeric constant
   45 | #define SD_CARD_TYPE_SD1      1
      |                               ^
c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src/SdCard/SdCardInfo.h:119:15: note: in expansion of macro 'SD_CARD_TYPE_SD1'
  119 | const uint8_t SD_CARD_TYPE_SD1 = 1;
      |               ^~~~~~~~~~~~~~~~
c:\Users\Admin\Documents\Arduino\libraries\STM32duino_STM32SD\src/Sd2Card.h:47:31: error: expected unqualified-id before numeric constant
   47 | #define SD_CARD_TYPE_SD2      2
      |                               ^
c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src/SdCard/SdCardInfo.h:121:15: note: in expansion of macro 'SD_CARD_TYPE_SD2'
  121 | const uint8_t SD_CARD_TYPE_SD2 = 2;
      |               ^~~~~~~~~~~~~~~~
c:\Users\Admin\Documents\Arduino\libraries\STM32duino_STM32SD\src/Sd2Card.h:49:31: error: expected unqualified-id before numeric constant
   49 | #define SD_CARD_TYPE_SDHC     3
      |                               ^
c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src/SdCard/SdCardInfo.h:123:15: note: in expansion of macro 'SD_CARD_TYPE_SDHC'
  123 | const uint8_t SD_CARD_TYPE_SDHC = 3;
      |               ^~~~~~~~~~~~~~~~~
In file included from c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src/ExFatLib/ExFatFile.h:34,
                 from c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src/ExFatLib/ExFatVolume.h:27,
                 from c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src/ExFatLib/ExFatLib.h:28,
                 from c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src/SdFat.h:30,
                 from c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src/SdFat_Adafruit_Fork.h:4,
                 from c:\Users\Admin\Documents\Arduino\libraries\Adafruit_SPIFlash\src/Adafruit_SPIFlash.h:32,
                 from C:\Users\Admin\Documents\Arduino\****t.ino:36:
C:/Users/Admin/Documents/Arduino/libraries/SdFat_-_Adafruit_Fork/src/common/FsApiConstants.h:72:15: error: redefinition of 'const uint8_t LS_DATE'
   72 | const uint8_t LS_DATE = 2;
      |               ^~~~~~~
c:\Users\Admin\Documents\Arduino\libraries\STM32duino_STM32SD\src/STM32SD.h:31:15: note: 'const uint8_t LS_DATE' previously defined here
   31 | uint8_t const LS_DATE = 1;
      |               ^~~~~~~
C:/Users/Admin/Documents/Arduino/libraries/SdFat_-_Adafruit_Fork/src/common/FsApiConstants.h:74:15: error: redefinition of 'const uint8_t LS_SIZE'
   74 | const uint8_t LS_SIZE = 4;
      |               ^~~~~~~
c:\Users\Admin\Documents\Arduino\libraries\STM32duino_STM32SD\src/STM32SD.h:33:15: note: 'const uint8_t LS_SIZE' previously defined here
   33 | uint8_t const LS_SIZE = 2;
      |               ^~~~~~~
C:/Users/Admin/Documents/Arduino/libraries/SdFat_-_Adafruit_Fork/src/common/FsApiConstants.h:76:15: error: redefinition of 'const uint8_t LS_R'
   76 | const uint8_t LS_R = 8;
      |               ^~~~
c:\Users\Admin\Documents\Arduino\libraries\STM32duino_STM32SD\src/STM32SD.h:35:15: note: 'const uint8_t LS_R' previously defined here
   35 | uint8_t const LS_R = 4;
      |               ^~~~
c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src/SdFat.h:455:16: error: conflicting declaration 'typedef class File32 File'
  455 | typedef File32 File;
      |                ^~~~
c:\Users\Admin\Documents\Arduino\libraries\STM32duino_STM32SD\src/STM32SD.h:37:7: note: previous declaration as 'class File'
   37 | class File : public Stream {
      |       ^~~~
exit status 1
`

acronis2000, 2025-07-20:
Hi fpistm!

I commented out the duplicate lines in the Adafruit library. Now I get these errors when I compile.

`c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src\ExFatLib\ExFatFilePrint.cpp: In member function 'bool ExFatFile::ls(print_t*, uint8_t, uint8_t)':
c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src\ExFatLib\ExFatFilePrint.cpp:71:19: error: 'LS_DATE' was not declared in this scope; did you mean 'FS_DATE'?
   71 |       if (flags & LS_DATE) {
      |                   ^~~~~~~
      |                   FS_DATE
c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src\ExFatLib\ExFatFilePrint.cpp:75:19: error: 'LS_SIZE' was not declared in this scope
   75 |       if (flags & LS_SIZE) {
      |                   ^~~~~~~
c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src\ExFatLib\ExFatFilePrint.cpp:85:20: error: 'LS_R' was not declared in this scope; did you mean 'LS_A'?
   85 |       if ((flags & LS_R) && file.isDir()) {
      |                    ^~~~
      |                    LS_A
exit status 1

Compilation error: exit status 1`

Can you tell me what I can try to do?

acronis2000, 2025-07-21:
I did as you told me. 
After entering the values in the code, new compilation errors appeared.

`c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src\SdCard\SdSpiCard.cpp: In member function 'bool SdSpiCard::begin(SdSpiConfig)':
c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src\SdCard\SdSpiCard.cpp:174:18: error: 'SD_CARD_TYPE_SD1' was not declared in this scope
  174 |       cardType = SD_CARD_TYPE_SD1;
      |                  ^~~~~~~~~~~~~~~~
c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src\SdCard\SdSpiCard.cpp:182:18: error: 'SD_CARD_TYPE_SD2' was not declared in this scope
  182 |       cardType = SD_CARD_TYPE_SD2;
      |                  ^~~~~~~~~~~~~~~~
c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src\SdCard\SdSpiCard.cpp:191:21: error: 'SD_CARD_TYPE_SD2' was not declared in this scope
  191 |   arg = cardType == SD_CARD_TYPE_SD2 ? 0X40000000 : 0;
      |                     ^~~~~~~~~~~~~~~~
c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src\SdCard\SdSpiCard.cpp:206:18: error: 'SD_CARD_TYPE_SDHC' was not declared in this scope
  206 |       cardType = SD_CARD_TYPE_SDHC;
      |                  ^~~~~~~~~~~~~~~~~
c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src\SdCard\SdSpiCard.cpp: In member function 'bool SdSpiCard::erase(uint32_t, uint32_t)':
c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src\SdCard\SdSpiCard.cpp:316:17: error: 'SD_CARD_TYPE_SDHC' was not declared in this scope
  316 |   if (type() != SD_CARD_TYPE_SDHC) {
      |                 ^~~~~~~~~~~~~~~~~
c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src\SdCard\SdSpiCard.cpp: In member function 'bool SdSpiCard::readStart(uint32_t)':
c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src\SdCard\SdSpiCard.cpp:514:17: error: 'SD_CARD_TYPE_SDHC' was not declared in this scope
  514 |   if (type() != SD_CARD_TYPE_SDHC) {
      |                 ^~~~~~~~~~~~~~~~~
c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src\SdCard\SdSpiCard.cpp: In member function 'virtual bool SdSpiCard::writeSector(uint32_t, const uint8_t*)':
c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src\SdCard\SdSpiCard.cpp:670:17: error: 'SD_CARD_TYPE_SDHC' was not declared in this scope
  670 |   if (type() != SD_CARD_TYPE_SDHC) {
      |                 ^~~~~~~~~~~~~~~~~
c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src\SdCard\SdSpiCard.cpp: In member function 'bool SdSpiCard::writeStart(uint32_t)':
c:\Users\Admin\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src\SdCard\SdSpiCard.cpp:735:17: error: 'SD_CARD_TYPE_SDHC' was not declared in this scope
  735 |   if (type() != SD_CARD_TYPE_SDHC) {
      |                 ^~~~~~~~~~~~~~~~~
exit status 1

Compilation error: exit status 1`

fpistm, 2025-07-22:
Welcome, hard to change this.
As stated before, like some are define and some are constant variables hard to properly handle this.
Moreover it will depends on the include ordering.
Those values are standard one defined by Arduino in their SD library:
https://github.com/arduino-libraries/SD/blob/c6352955f8c250a2330a7f5d9c2d34484b46cc8b/src/utility/Sd2Card.h#L166-L170
https://github.com/arduino-libraries/SD/blob/c6352955f8c250a2330a7f5d9c2d34484b46cc8b/src/utility/SdFat.h#L57-L62

acronis2000, 2025-07-22:
I got it. In any case, you helped me solve the problem and it works well! Thank you again!