stm32duino/STM32SD #86: Library breaks Hardware timer interrupts

lekrom, 2025-09-10:
**Please, before reporting any issue**
- Make sure you are using the latest Arduino_Core_STM32 version.
  https://github.com/stm32duino/Arduino_Core_STM32/releases/latest
- Make sure the issue is not already reported/fixed on GitHub or discussed on the [stm32duino forum](https://stm32duino.com)
- Submit a GitHub issue only for reporting a problem related to the repository.
- **Avoid** to submit a GitHub issue for project troubleshooting.

Any questions/feedback/suggestions should be discussed on the [stm32duino forum](https://stm32duino.com):
  * questions on the [STM32 Core](https://www.stm32duino.com/viewforum.php?f=35)
  * bugs/enhancements on the [STM core: Bugs and enhancements](https://www.stm32duino.com/viewforum.php?f=38)

&#x26A0; When reporting any issue, please try to provide all relevant information to help on its resolution.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Complete source code which can be used to reproduce the issue. Please try to be as generic as possible (no extra code, extra hardware,...)

Steps to reproduce the behavior:
1. Plug
2. Press USER_BTN
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows]
 - Arduino IDE version: [e.g. 1.8.8]
 - STM32 core version: [e.g. 1.4.0]
 - Library version: [e.g. 1.2.0]
 - Tools menu settings if not the default: [e.g. Newlib Standard, No Serial]
 - Upload method: [e.g. SWD]

**Hardware (please complete the following information):**
 - Board Name: [e.g. Nucleo F429ZI]
 - Hardware Revision: [e.g. Rev B]
 - Extra hardware used if any: [e.g. X-Nucleo IKS01A1]

**Additional context**
Add any other context about the problem here.

lekrom, 2025-09-10:
Hi, 
I am developing a custom board using the STM32F413RG. This board uses SPI peripherals, CAN, an SD card (for logging) via SDIO as well as a frequency counter input using the HardwareTimer. I used this example sketch successfully on my hardware to get the HardwareTimer going: https://github.com/stm32duino/STM32Examples/blob/main/examples/Peripherals/HardwareTimer/InputCapture/InputCapture.ino -> I am using TIM2 on PORTA,1

Upon implementing this in my larger program, the hardware timer interrupts (as per the example I mentioned) no longer worked. After extensive troubleshooting, I found that using the STM32SD library breaks the hardware timer interrupts. Even simply including this library (not actually calling any functions from it) on the bare example sketch above, breaks the interrupts. When debugging, I can see that TMR2 is running however.

Can someone please make sense of this?

Regards,
Lekrom

fpistm, 2025-09-10:
No information, only PR template.

lekrom, 2025-09-10:
Hi, Thanks for the reply. 
I am not using the same pins.
For input capture, i use PA1 (not used by SDIO on the F413RGT6)
For SDIO I use:
D0 PC8
D1 PC9
D2 PC10
D3 PC11
CMD PD2
CLK PC12

I cant see why they interfere either, but they do, as soon as I comment out the #include <STM32SD.h>, it works again.

lekrom, 2025-09-10:
My full app is less than 10% of the available flash.
I use platformio, using the standard STM32F413RGT6 linker script from CubeIDE.
I need to use a custom linker script as I also use a bootloader and thus specify flash offsets.

While testing at the moment, the bootloader is not used and so I use the unmodified (no offset) linker script.

I started disabling chunks of of my program to figure our where the conflict was. As soon as I killed the SD related code (and the #include) it started working again. I then went back to the example sketch and simply added the #include to test, that broke then also. (Example sketch uses 3% of flash)

I know it makes little sense, I will keep fiddling...

-Lekrom